prusa3d/PrusaSlicer #4475: [Feature request] Configurable scale size per filament (ABS shrinks!)

rackley096794, 2020-07-04:
### Version
2.2.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
MK3S

### Behavior
* Some materials shrink when cooled, so they need to **always** be scaled at a certain percentage for them to print with the as-designed dimensions.  ABS, ASA, and various blends of these materials are affected and wind up smaller than their as-designed dimensions after cooling.
* Some people report needing to upscale by 2-3%, some more.  Personally I need to scale at 101% with the FR-ABS blend I print lots of parts with.  
* However, trying to remember to do this every time you print a part (and for every part printed on a platter!) is very error prone.  
* This is a real problem when printing parts that mate together and larger parts where a small scale percentage difference becomes amplified enough that things don't fit anymore.  For example here is a fan mount I regularly print for a commercial product.  If I print it at 100% scale instead of 101%, the mount holes don't line up for the fan and the case, resulting in 9 hours of wasted print.  I realize this can be mitigated by saving a reference print as a .3mf, but with lots of different parts that constantly evolve, this isn't really a good workaround.  Plus, every time you add a new part it scales at 100%, adding in another new pitfall to stumble into.

[image]

_Is this a new feature request?_  Yes

DrKropp, 2020-07-22:
I would love this as an option. Recently printed 30 copies of a part in ABS and forgot to add my normal 2.5% scale factor. Wasted almost 2 kg of filament and 4 days of print time on that error.

MStylesMS, 2020-10-30:
I agree with this request.  Having a default shrinkage factor for each filament profile based upon the middle range of the range of the printing temperature vs a typical ambient temperature (say 22.5C).  This would make sense one the Advanced page of the filament definition.

golfromeo-fr, 2020-10-31:
The feature has been put in Cura many months ago according to a famous 3D printing youtuber but Cura is so complicated that I cannot find the setting yet.

It is not only a problem with ABS, PLA is prone to shrinkage but less, like the need to scale up 100.5%

rackley096794, 2020-11-05:
I'm working on an implementation of this...  Will submit a PR when complete.

neophyl, 2020-11-05:
Its also currently in Super Slicer.  Its been added to the filament profiles.

manolitto, 2021-01-13:
I think this is more or less a duplicate of #3078

Regarding SuperSlicer: The solution they have is suboptimal IMHO, they only compensate in x and y direction.
See my [other comment](https://github.com/prusa3d/PrusaSlicer/issues/3078#issuecomment-759826525)

rackley096794, 2021-01-14:
I stopped working on this feature.  I was making good progress, but I noticed my super simple [update](https://github.com/prusa3d/PrusaSlicer/pull/5069/) to the Getting Started docs was languishing in neglect-land (and still is over two months later).  Quite frankly I don't want to waste my time and write a real PRs for a software package that will never be looked at.  😞

manolitto, 2022-03-24:
> I think this is more or less a duplicate of #3078
> 
> Regarding SuperSlicer: The solution they have is suboptimal IMHO, they only compensate in x and y direction. See my [other comment](https://github.com/prusa3d/PrusaSlicer/issues/3078#issuecomment-759826525)

I learned that SuperSlicer is doing right by just compensating in x and y. Reason is: the z size is given by the vertical steps which are always exact. So, every single layer is printed exactly at the height it should be - regardless of the shrinking of already printed layers.

Gefionious, 2022-03-31:
I should note, that it appears there are work-arounds available... I am looking at this: [https://www.thingiverse.com/thing:2484766#Summary]
I think it is very possible to add g-code via the M92 steps per mm correction in a script, perhaps even checking the filament profile. Something to consider.

stepikovo, 2022-04-13:
I'm closing this as a duplicate of #3078

rtyr, 2022-05-22:
Reopened. I don't see it as a duplicate of 3078.

dbech, 2023-01-30:
seems like a duplicate of #3078, so much so when this issue is referenced in discussion only they are both listed. Just merge em.

RudolphRiedel, 2023-02-04:
I do not agree either that this here is a duplicate of #3078 either since adding a plain scale factor to the filament settings that used when loading an object is really not the same than "have a function to scale your prints in dependancy of the CTE of the material and the heatbed temperature".
I would like to have this for functional parts in PETG for which I need to scale prints up to 100.5% in X/Y/Z, at least with the PETG that I am using right now.

dbech, 2023-02-07:
if you shrinkage truely is a perfect 0.5% then scaling based on temp will get you the same result.

manolitto, 2023-02-07:
Please keep in mind that scaling in Z is generally not necessary (or needs a different factor). This is due to the fact that while the material shrinks during printing, the printer still does print the next layer at exactly the perfect height. Thus the Z-shrinkage is compensated automatically in most cases.

RudolphRiedel, 2023-02-07:
It is not perfectly 0.5% but close enough for the filament I tested, 0.6% might be more accurate.
And the Z for what I am printing is not high enough to make a real difference, my object is 250 mm x 150 mm x 20 mm.  
If it would shrink during printing though, which I am sceptical about, it would lead to massive issues the higher the print is.

bszonye, 2023-02-11:
The filament shrinks any time the temperature drops, which first happens during printing as the filament drops from nozzle temperature to bed temperature. However, because each layer is stuck down to the one below it, down to the build plate, the shrinkage tends to create more tension than actual lateral motion. That tension is why corners will curl up from the build plate if they aren't stuck down well enough.

The physics of the Z dimension are a bit different, although the end result is similar. As the first layer cools to bed temperature, you get tension in the X & Y dimensions and shrinkage in the Z dimension. However, that won't affect the height of the _next_ layer. The nozzle squashes filament into the layers below, which always leaves the top of the new layer at nozzle height exactly. Any variation in the bottom layers will only change the amount of squashing. (This is the effect that @manolitto described, in more detail.)

That said, I think the SuperSlicer devs oversimplified the differences between the XY and Z dimensions. I suspect that they're overestimating how much XY shrinks during the print and underestimating how much Z shrinks after printing.

I've had good results with the calibrator that @Gefionious used above, which makes some very different assumptions. For example, it assumes that the whole part ends the print at bed temperature and shrinks uniformly from there to room temperature. That's also a bit of an oversimplification, but it works better for me than assuming that Z doesn't shrink at all.

johnkray, 2023-03-07:
Chiming in to show my support for getting a shrinkage compensation feature added to the filament settings in Prusa Slicer. I work for a small 3D print service and this is a constant problem with our ABS parts.

RWL-69, 2023-05-31:
This is a feature that is needed.  Especially if you are printing the same object with different filaments.  Or printing a mesh model file that you have no control over.

cmdremily, 2023-06-16:
I would very much like this as well.

gsasha, 2024-06-18:
This is the primary feature that makes me use Orca Slicer, while I'd very much prefer Prusa Slicer.

rtyr, 2024-06-18:
Implemented in 2.8.0-alpha5.

closed 2024-06-18 by rtyr

johnkray, 2024-06-18:
> Implemented in 2.8.0-alpha5.

Yes, thank you!

RudolphRiedel, 2024-06-23:
Really like the change.
However, there is no indication whatsoever that the setting actually does anything?
There is a noticeable difference in size when measuring the object on screen in the preview when setting "Shrinkage compensation XC" to the maximum of 10% so it appears to be working fine, but there are no numbers presented to go with this.