prusa3d/PrusaSlicer #4477: [featurerequest] Custom printing patterns for bottom and top surface

pedjas, 2020-07-05:
### Behavior
Bottom surface is always printed with lines angled 45 degrees to the right. For top surface it depends ho high it is, it may be angled 45 degrees to the right or to the left.

It would be great to have choice of printing patterns for bottom and top surface, like zigzag, sinusoidal lines, chessboard, circles or similar. That would help having interesting surfaces even when printing on plain sheet.

_Is this a new feature request?_
Yes

n8bot, 2020-07-05:
Is this feature not already present?
[image]
Or are you looking for more patterns?

rtyr, 2020-08-19:
Separate fill pattern for top/bottom is already implemented. Closing.

closed 2020-08-19 by rtyr