prusa3d/PrusaSlicer #4478: Pause Print feature ignores G-Code Flavor

fake-name, 2020-07-05:
### Version
2.2.0+win64

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)

Custom based on Duet3D Duet 2

### Behavior

Currently, if you are trying to pause at a specific layer, the pause dialog **always** inserts a `M601` command. 

From what I can tell, basically no one other then prusa printers support `M601`.

Firmwares that use `M25`:

 - Marlin
 - Smoothie
 - Repetier

Firmwares that use `M226`:

 - RepRapFirmware
 - Teacup

Firmwares that use `M1`:

 - LinuxCNC 
 - machinekit

Firmware where I can't determine:

 - Sailfish
 - Makerware

This seems like it's basically bound to the g-code flavor, so it should be modified by the selected type. I was quite surprised the first time I ran a file with pauses and my printer never stopped.

n8bot, 2020-07-05:
I agree, there are some annoying choices for Gcodes. I would like the ability to define the Gcodes for the "flavour" ourselves. It should be pretty simple to allow us to customize these gcodes.

That said, on the Duet, you can simply place a file called "M601.g" into your sys/ folder, and that gcode file will be run when M601 is called. You could call M600 with M601.
```
; Contents of file M601.g
M600
```
Awkward, I know...

bubnikv, 2020-10-28:
In PrusaSlicer 2.3.0-alpha2 we have implemented custom G-code for pause print events.
Does that solve your issue?

GoodyGizmos, 2021-02-12:
> In PrusaSlicer 2.3.0-alpha2 we have implemented custom G-code for pause print events.
> Does that solve your issue?

Not for me, because I print through OctoPrint which only recognized M25, not M601 (at least not reliably)

bubnikv, 2021-02-15:
[image]

closed 2021-02-15 by bubnikv