prusa3d/PrusaSlicer #4484: [Feature Request] Adjusted final layer height for correct model height

The3Designer, 2020-07-05:
### Version
2.2.0

### Operating system type + version
Windows 10 - Version 2004

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S with 3.9.0 firmware version

### Behavior
When slicing a model with a layer height that does not perfectly "fit" in the model's height, the object becomes taller (or smaller, depending on how tall the model exactly is). The example file I've included is 10.0mm tall. When slicing it with a 0.3mm layer height (and a first layer height of 0.2mm, which is normal for every layer height), the model becomes 10.1mm tall.

What I would suggest/like, is that the final layer of the print is adjusted when necessary, if the used layer height does not allow for a matched model height. Thus, when I slice the example model with a 0.3mm layer height, I want/expect both the first and final layer height to be 0.2mm.

#### Project File (.3MF) where problem occurs
The image below shows that the model ends up being 10.1mm tall, instead of the 10.0mm of the original/actual CAD model.

[image]

I've saved the project as a .3MF file and included it with this post.
[error_example.zip](https://github.com/prusa3d/PrusaSlicer/files/4875111/error_example.zip)

n8bot, 2020-07-05:
If this feature is implemented, it would need to be an option as having this as default behavior would be :banana::banana:.

That said, this can currently be achieved by manually adjusting the layer height for the final layers, using the height range modifiers.

The3Designer, 2020-07-05:
> 
> 
> If this feature is implemented, it would need to be an option as having this as default behavior would be 🍌🍌.
> 
> That said, this can currently be achieved by manually adjusting the layer height for the final layers, using the height range modifiers.

Agreed, I would definitely turn this feature on as default behaviour. In fact, I think it should be the default behaviour when implemented, with an option to turn it off and maintain a constant layer height. Reasoning behind this is that I believe that object (dimension) accuracy is more important than having a constant layer height.

Thanks for the height range modifiers tip, I didn't know that this function existed, but it works like a charm :)

n8bot, 2020-07-05:
> [...]
> 
> Reasoning behind this is that I believe that object (dimension) accuracy is more important than having a constant layer height.
> 

Good point. And actually, this could safely be default behaviour in conjunction with the current ability to set min and max layer heights for variable layer heights. The min max there can be respected, and then there is no risk of having an oddly sized layer added on top. Or at least the risk is minimized. My only concern would ever be having a odd size layer like .09345 mm or something, or a layer that is much too small like 0.009 mm or something silly.

> Thanks for the height range modifiers tip, I didn't know that this function existed, but it works like a charm :)

This feature is definitely well-hidden. I had to go digging to find it one day. Many users aren't aware of it, so I'm glad I could help in that regard.

ochm, 2020-07-06:
Why do not you adjust the height of first layer?

The3Designer, 2020-07-06:
> 
> 
> Why do not you adjust the height of first layer?

Because PrusaSlicer uses a 0.2mm first layer height, regardless of the layer height for the rest of the print. If this is default behaviour, then I expect it to be required for proper bed adhesion. Furthermore, If I wanted the model to be exactly 10.0mm, I would have to change the first layer height to 0.1mm, which is a much larger difference compared to the 0.3mm layer height than two layers (first and final) of 0.2mm.

closed 2026-09-29 by prusaslicer-bot[bot]