prusa3d/PrusaSlicer #4487: Prusaslicer 2.2.0 freezes the UI for several seconds frequently at unpredictable times.

BillWhite, 2020-07-05:
### Version
PrusaSlicer-2.2.0+linux-x64-202003211856.AppImage, which I just today installed from your web site.  This also happens with my locally built version of HEAD from github (commit d8a81e978a8e95d16ec269a77be37dabda6cccff).

### Operating system type + version
Ubuntu Studio Linux 19.10, 

System Info:
PrusaSlicer
Version:   2.2.0+linux-x64
Build:     PrusaSlicer-2.2.0+linux-x64-202003211856

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 5.3.0-62-lowlatency x86_64
Total RAM size [MB]: 67,296MB
OpenGL installation
GL version:   4.6.0 NVIDIA 440.100
Vendor:       NVIDIA Corporation
Renderer:     Quadro M1000M/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
[... 317 lines cut ...]
This makes prusaslicer entirely unusable to me.

### Behavior
The mouse pointer freezes frequently but unpredictably when the the pointer goes into the initial configuration dialog, or when the pointer goes into the plater tab.

* _Steps needed to reproduce the problem_
1. Delete ~/.Prusa* to delete old configurations.
1. Run the appimage file, with suitable permission changes. (The same thing happens with a locally built version, but I'll not repeat this.)
1. Emulate the patience of Qui-Gon Jinn while waiting for the mouse cursor to unfreeze while answering the configuration questions. Note that when the pointer is frozen, the window manager commands, such as ALT-TAB to select other windows or CTRL-ALT-LEFTARROW to move to another workspace, are not responsive until the pointer is unfrozen.
1. When the initial configuration is done, and the cursor is in the Plater tab, move it to the tabs area above the Plater tab. The cursor may freeze or not, depending on how fortunate you are.
1. In the tabs area, select one of the settings tabs, say Printer Settings. Move the cursor around with mad abandon, and notice that no freezing occurs. This is true for all the settings windows.
1. Select the Plater tab and move the mouse until it freezes.
1. Note that I have wxWidgets 3.0 not 3.1. Since I am slightly behind the curve, still on U19.10, I don't have wxWidgets 3.1 easily at hand. I'm about to upgrade, and will add a note if that makes a difference.

* _Expected Results_
A responsive cursor.

* _Actual Results_
A frequently frozen cursor.

_Is this a new feature request?_
No.

#### Project File (.3MF) where problem occurs
I don't actually have a project file. I haven't loaded any models yet.

BillWhite, 2020-07-10:
This seems to be related to the graphics card. My computer has an nVidia card and an intel card. When the nVidia card is activated I encounter this problem. When the intel card is activated the problem goes away. The nVidia drivers, both from nVidia and the nouveau driver, are frequently problematic, so it's not really astonishing.

Thanks, but I think I'm good now.

UltraSalem, 2020-07-12:
I've also been very rarely getting random pauses/lock ups in Prusa Slicer, on pop_os 20.04, using an AMD graphics card/cpu.

unrepentantgeek, 2020-08-01:
This occurs to me as well (and has for several versions of PrusaSlicer)  It clears up if I'm not using the nvidia card either due to having crashed it or disabled it.  I don't see any messages in dmesg.  The entire screen locks up (videos stop playing but audio continues)

I'm running the official Nvidia driver version 440, but this has happened for a few versions now.  Machine is a Dell XPS15 with Nvidia GTX1050 and Intel video.  Happy to provide all the debugging info you want, and can reproduce easily.

bubnikv, 2021-09-25:
I am afraid we cannot help. Even if there was some workaround possible at PrusaSlicer side, we would have to get our hands on such a computer.

Last fall I was installing Linux on a handful of old laptops to be donated to local schools for remoting. OpenGL support was really sketchy.

closed 2021-09-25 by bubnikv