prusa3d/PrusaSlicer #4489: Configurable Modifiers

RGol2020, 2020-07-06:
### Version
2.2.0 win 64

### Operating system type + version
Windows 10 pro

### 3D printer brand / version + firmware version (if known)
Prusa mk3s

### Behavior
New feature request

I have used modifiers to ensure that parts of the print are solid (it's an end cap on a beam which needs to have the edges of the end face trimmed to the beam on site).

I have done this using rectangular box modifiers, which is 'clunky'.

Would it be possible to define a custom shape for the modifier?

#### Project File (.3MF) where problem occurs
[image]

neophyl, 2020-07-06:
You can load other stl files in as modifier parts already. So just create the shape you want and add it as a modifier.

RGol2020, 2020-07-06:
OK thanks, I was not aware. 

 

Should I close the thread?

 

Regards

 

Richard

closed 2020-07-06 by RGol2020