prusa3d/PrusaSlicer #4491: Same settings, different models, different speed results

gnydick, 2020-07-06:
### Version
2.2.0+win64

### Operating system type + version
Windows 10 2004

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Using the same settings (which will be attached), slicing two different models yields extremely different speed assignments.

With the exception of the first layer speed, the speed settings are all set to *0* with a maximum print speed set. Filament auto-cooling is disabled and volumetric flow isn't a limiting factor

When sliced, the first model (articulated-dish) is almost entirely the max speed. The second model is much, much slower. Can't really figure out why this is.

Slicing them both together yields slower speed for the first model.

#### Project File (.3MF) where problem occurs

[files.zip](https://github.com/prusa3d/PrusaSlicer/files/4880413/files.zip)

rtyr, 2020-07-14:
I don't recommend using the autospeed feature. It doesn't always work realiably and it is probably still considered as "experimental" feature. We may remove this inherited feature in the future.

gnydick, 2020-07-14:
Please don't remove this!?!?! I have a very fast printer that can print at
high quality at speed. It is much more convenient to give it a max speed
and let the other settings alter it.

Really, why is it so difficult to have individual speed settings per
feature type and have auto cooling and volumetric rate override them vs.
having a default speed that fills in the other boxes and does the same
thing?

VanessaE, 2020-11-01:
NO, DO NOT REMOVE!  

Some hotends work better with autospeed.  If it has problems, FIX them.

closed 2026-09-29 by prusaslicer-bot[bot]