prusa3d/PrusaSlicer #4493: Feature request: Dark mode on Windows (10)

Sundheden, 2020-07-07:
I would love to see the dark version available on Mac, also be available on windows! 

Reference:
[https://i.imgur.com/HsT7M6T.png](https://i.imgur.com/HsT7M6T.png)

n8bot, 2020-07-11:
This was discussed on one of the prusa live streams. I believe they mentioned that it was difficult “due to windows API” or something. 

I agree though, it’s worth figuring out!

MikeRatcliffe, 2020-08-31:
I really want this too as I have to use a dark theme for accessibility reasons.

I was considering implementing this myself but the win32 dark theme API is undocumented and could possibly change. It's not likely to change though considering MS has updated most of it's apps to use it.

See https://github.com/ysc3839/win32-darkmode for an example of how to use the current API.

bubnikv, 2020-09-01:
If you want a dark mode in PrusaSlicer on Windows 10, you should post a
complaint using the Windows 10 feedback application "Feedback Hub". If you
open an issue there, please post its name, so others could upvote it.

bubnikv, 2020-09-01:
> See https://github.com/ysc3839/win32-darkmode for an example of how to
use the current API.

@yusanka tried that quickly, but the Win32 dark mode support is half baked
(they implemented just what they needed to support Windows Explorer and
their other Win32 based system applications), so we don't think it is worth
the effort as of now. We need more love from Microsoft.

MikeRatcliffe, 2020-09-01:
> @YuSanka tried that quickly, but the Win32 dark mode support is half baked (they implemented just what they needed to support Windows Explorer and their other Win32 based system applications), so we don't think it is worth the effort as of now. We need more love from Microsoft.

I thought somebody had probably tried.

> If you want a dark mode in PrusaSlicer on Windows 10, you should post a complaint using the Windows 10 feedback application "Feedback Hub". If you open an issue there, please post its name, so others could upvote it.

@bubnikv I have logged an issue in the Microsoft feedback hub [here - Win32 dark mode API too basic to use in the wild](https://aka.ms/AA9jecu).

@YuSanka Can you let the Microsoft devs know which API functionality you need in order to implement this at https://aka.ms/AA9jecu?

@EverybodyElse If you would like dark mode functionality in PrusaSlicer please upvote this issue [here - Win32 dark mode API too basic to use in the wild](https://aka.ms/AA9jecu).

Kollimolli, 2020-10-10:
For me it looks like a Prusa devs are bit lazy when blaming Microsoft about that API. It is not that hard to create an option for a dark theme. Even a temporary dark theme would be nice until Microsoft fixes their dark mode APIs, if that never happens.

phazei, 2020-10-29:
Plenty of apps have lots of color themes.  Dark mode in windows has been around for ages, it doesn't seem likely that MS is going to update the API for general use.  There are also plenty of windows apps that are dark themed, so perhaps some consideration could be put into this under the assumption that the API won't be updated.

bubnikv, 2020-10-29:
> @Kollimolli
> For me it looks like a Prusa devs are bit lazy when blaming Microsoft about that API.

Exactly.

> It is not that hard to create an option for a dark theme.

Yeah, it is trivial. We are just lazy. Please provide a pull request.

@phazei

> Plenty of apps have lots of color themes. Dark mode in windows has been around for ages, it doesn't seem likely that MS is going to update the API for general use. There are also plenty of windows [...]

It depends on the UI toolkit your application is built upon. Win32 does not support dark mode. Some others do. If the UI toolkit supports the dark theme, then it is quite easy to just skin the UI. If it does not, then it is very difficult.

bubnikv, 2020-11-23:
https://github.com/ysc3839/win32-darkmode
This is the only support we have for dark mode, which is 
1) a hack depending on unpublished Microsoft internal API which may break at any time
2) the last time @YuSanka tried that it was very incomplete for our PrusaSlicer purpose and we still hoped that Microsoft will provide an official API. It is nearly two years now since Microsoft released the first dark mode settings in Windows 10, but there is still no developer support for dark mode.

One may check the commit log of the unofficial Win32 dark mode support project how fragile the support is and that it may break with any Windows release.
[image]

Kollimolli, 2020-11-25:
I understand this. It's shame that Microsoft have almost abandoned that API.

What i don't understand is this: Prusa devs has been waited long time and still waiting with fingers crossed. Maybe Microsoft never fixes dark mode APIs. Perhaps it is better develop an alternative way to make dark mode possible? Currently i have installed many apps that have dark mode/theme and all them works well, so there must be an another way to make dark mode/theme.

"Don’t sit and wait forever for the mountains to move out of the way. Use a detour"

bubnikv, 2020-11-26:
> What i don't understand is this: Prusa devs has been waited long time and
still waiting with fingers crossed.

We are certainly not waiting with our fingers crossed. Trust me, we are all
very busy.

An alternative solution is to either abandon the Win32 API, which means an
extreme amount of time and effort, or hacking around the Win32 limitations,
which is also time consuming and the result will be fragile against changes
at the Microsoft side. There is no easy, cheap and quick solution. And if
we are investing our time into the dark mode support, we will not be
investing our time into making the slicer produce better supports, faster
prints etc.

Kollimolli, 2020-11-26:
So long answer in short: Prusa devs still continues waiting for forever if Microsoft someday fixes dark mode APIs and until that there will not be a dark mode because there is no easy or cheap way to do it without MS dark mode APIs? 

Please dont understand me wrong, i appreciate lot your efforts and work making Prusaslicer better and better, it is already one of the best slicers available. But there is many users, me too, who cannot use or it is hard to use apps with light theme because visual impairment or other reasons. So it is not just a matter of taste, it is thing that defines can app be used or not. For me this is more important thing than better supports or faster prints.

MikeRatcliffe, 2020-11-26:
> So long answer in short: Prusa devs still continues waiting for forever if Microsoft someday fixes dark mode APIs and until that there will not be a dark mode because there is no easy or cheap way [...]

I am in the exact same position... I am very sensitive to light and it basically results in the same symptoms as a stroke and seizures (I recover in around 3 days).

Sadly, the accessibility settings on Windows are too harsh so they don't work for me either.

Even inverting the colors would be a start but at the same time I understand that we are in the minority so I understand why they focus on functionality above accessibility.

SM-26, 2020-12-24:
I might be missing something, but I hope this helps:

My first thought was just using the registry:
HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Themes\Personalize
something like this for java:
``` java
public static boolean isSystemDarkTheme () {
	boolean isDarkTheme = false;
	if (OS.WIN32_VERSION >= OS.VERSION (10, 0)) {
		TCHAR key = new TCHAR (0, "Software\\Microsoft\\Windows\\CurrentVersion\\Themes\\Personalize", true); //$NON-NLS-1$
		long [] phkResult = new long [1];
		boolean regKeyFound = false;
		if (OS.RegOpenKeyEx(OS.HKEY_CURRENT_USER, key, 0, OS.KEY_READ, phkResult) == 0) {
			regKeyFound = true;
		} else if (OS.RegOpenKeyEx(OS.HKEY_LOCAL_MACHINE, key, 0, OS.KEY_READ, phkResult) == 0) {
			// Try reading from HKLM
			regKeyFound = true;
		}
		if (regKeyFound) {
			int [] lpcbData = new int [1];
			TCHAR buffer = new TCHAR (0, "AppsUseLightTheme", true); //$NON-NLS-1$
			int result = OS.RegQueryValueEx (phkResult [0], buffer, 0, new int[] {OS.REG_DWORD}, (TCHAR)null, lpcbData);
			if (result == 0) {
				int[] lpData = new int[lpcbData[0] / TCHAR.sizeof];
				result = OS.RegQueryValueEx(phkResult[0], buffer, 0, new int[] {OS.REG_DWORD}, lpData, lpcbData);
				if (result == 0) {
					isDarkTheme = (lpData[0] == 0);
				}
			}
			OS.RegCloseKey (phkResult [0]);
		}
	}
	return isDarkTheme;
}
```

or maybe use parts of this example in Python?
``` python
 try:
            import winreg
            with winreg.ConnectRegistry(
                None,
                winreg.HKEY_CURRENT_USER
            ) as registry:
                with winreg.OpenKey(
                    registry,
                    'SOFTWARE\\Microsoft\\Windows\\CurrentVersion\\Themes\\Personalize'
                ) as key:
                    value, value_type = winreg.QueryValueEx(key, 'AppsUseLightTheme')
        except Exception as e:
            print('AutoDark: Error(_is_dark_os("windows")): ', e)
```
I know it's not "the right way", but it could work, right?

if not, maybe importing the WindowsApp.lib ? If I understand it correctly, it could help us do what UWP can do(?)

> The Microsoft.Windows.SDK.Contracts NuGet package gives .NET Framework 4.5+ and .NET Core 3.0+ applications access to Windows 10 WinRT APIs, including Windows.UI.ViewManagement.Settings

some C example: https://github.com/ysc3839/VCMPBrowser/blob/darkmode/DarkMode.h

bubnikv, 2021-01-05:
@SM-26 If you think that is sufficient, then please test it and provide a pull request. IMHO it is just around 3% of the total effort, the biggest issue will be to skin the Win32 controls, that Microsoft still did not bend to support the dark mode.

johny-mnemonic, 2021-03-19:
I am a sysadmin, not an app developer, so I may be completely wrong here, but I will ask anyway.

Why on earth is this project still maintaining the branch of the application that relies on that pile of obsolete junk called Win32 API? :-O

I was living with an impression that only old obsolete/unmaintained apps still use that thing and everyone else already moved on. So to find out, that such a great, thriving and active project like Prusa Slicer still maintains a branch which relies on Win32 API doesn't make any sense to me.
You obviously have Linux and Mac versions of your app, which are with almost 100% certainty using some different (hopefully multiplatform) API and not Win32, so why are you still dragging this ball tightened to your leg? :-O
This looks to me like a self-inflicted wound that you are keeping open and bleeding.

Not being able to easily implement dark skin to your app must be just an tip of the iceberg of the issues that this relying on the old unmaintained/undocumented MS API brings...

bubnikv, 2021-03-19:
PrusaSlicer is built upon wxWidgets toolkit, which uses Win32 API behind
the scenes.

> Why on earth is this project still maintaining the branch of the
application that relies on that pile of obsolete junk called Win32 API? :-O

Chrome (and Edge) browsers use Win32 API on Windows. File explorer, notepad
and all the rest of the "old" windows applications are still built using
Win32 API.

johny-mnemonic, 2021-03-19:
@bubnikv Chromium based browsers use Win32 API on Windows? :-O That probably shows my lack of knowledge ;-) But also brings important question:
How come these do support Windows 10 Dark mode while you are saying it is not possible with the current state of Win32 API?

lukasmatena, 2021-03-20:
@johny-mnemonic I will try to describe the issue in more detail, there clearly are misunderstandings. The "I don't know, but it can't be difficult because I saw a dark button in application XYZ" opinion simplifies things a bit too much. @bubnikv will certainly correct me if needed.

1. Mocking Windows applications for relying on Win32 API is a bit odd. Win32 API is widely used by many applications and unlike your claims, it is still developed and it is documented. The fact that Microsoft themselves still heavily use this "pile of junk" for their applications also says something.

2. Win32 API is usually not called directly, but it is hidden under some wrapper library (which may or may not be multiplatform). Examples include MFC, .NET, wxWidgets, Qt, FLTK and many others. Your impression that "everyone moved on" is not precise. Most moved away from the core and forgot that it is there.

3. Win32 API does not oficially support dark mode. Not in the sense that it would allow to turn it on and controls would magically start rendering dark. However, it allows (and has allowed for at least 20 years) to override rendering methods of the individual controls and draw whatever the hell you want. I myself did it this way 15 years ago when Windows XP was a thing, long before dark modes. It is simple if we are talking about one button, gets complicated with menus, title bars etc.

4. So, when it comes to rendering controls, there are two approaches that the wrapper libraries can take. One is to ask Windows to draw the controls (wxWidgets), the other is to take over and draw it themselves (Qt, FLTK). The former approach is less flexible, the latter is more work but in principle allows the authors to implement dark mode themselves, without any extra support.

So, technically there are several options how to get dark mode into PrusaSlicer:
a) MS extends the API, then we can "just turn it on"

b) wxWidgets would change the way it renders controls, implements both light and dark mode, we "just turn it on" (not going to happen)

c) we switch to custom rendering of controls on Windows ourselves - lot of work and a maintanence burden (you're talking about being multiplatform, but this is exact opposite)

d) we replace wxWidgets with another library whose authors already did the hard work - this is LOT of work and has many many more consequences. If it ever happens, it will not be because of Win dark mode.

The choice is not simple. Especially when you think about all the other features we will not have time to work on because of this.

johny-mnemonic, 2021-03-20:
Thanks for the detailed explanation @lukasmatena!

So you are using wxWidgets also on Mac and Linux, just calling another underlying OS specific API?
If not and you're already using something like QT on those OSes, then what is the reason not to use the same thing on Windows as well?

n8bot, 2021-03-20:
> Thanks for the detailed explanation @lukasmatena!
> 
> So you are using wxWidgets also on Mac and Linux, just calling another underlying OS specific API?
> If not and you're already using something like QT on those OSes, then what is the reason not to use the same thing on Windows as well?

wxWidgets is multiplatform. It handles everything for macOS, Windows, and Linux. All PrusaSlicer has to do is implement wxWidgets into the app, and the "translation" to the different platforms is taken care of by wxWidgets.

MLiv79, 2021-04-03:
[image]

Now seems to be there the options to have a dark gui but i don't know how to apply, i've tried to change values without results

klimlik, 2021-04-05:
It looks like there is now a solution for users who uses Windows high contrast-themes.

https://github.com/prusa3d/PrusaSlicer/issues/3038

bubnikv, 2021-04-07:
Sent to wx-users list (the list for discussing usage and issues of the multi-platform user interface library that PrusaSlicer is built upon):

> It is known that Microsoft did not publish their Win32 dark mode API, most likely namely because it is far from complete. There is no information from Microsoft about the time schedule of the Win32 dark mode API or whether they will publish Win32 dark mode API at all. I am getting a feeling that Microsoft is ditching Win32 API in favor of the UWP API and the dark mode will never come to Win32 API.

> So Microsoft implemented the dark mode for those Win32 controls that are used by their products. Somebody made an effort some two years ago to reverse engineer that API, however that effort stopped about a year and half ago.

> https://github.com/win32-darkmode/win32-darkmode

> Luckily, there is an effort on the famous Notepad++ editor.

> Dark Mode begins! by adzm · Pull Request #9587 · notepad-plus-plus/notepad-plus-plus (github.com)

> Notepad++ is a pure Win32 application, thus the Dark Mode effort is not clouded by any multi-platform layers. The Notepad++ project is active and widely used (I am using it every day myself), so there is quite a chance that they will polish the Dark Mode at least for their usage.

> We at PrusaSlicer are asked quite often for Dark Mode support on Windows. We were waiting for Microsoft to help, but that will likely not happen. In the meanwhile, our customers and non-customers (users of 3rd party Chinese 3D printers) blame us for being lazy not implementing the dark mode. Thus we may invest quite a lot of time to port the dark mode from Notepad++ to PrusaSlicer. However it looks like it will be much easier to integrate it into wxWidgets.

> That's the heads up on our Dark Mode troubles and effort. We hope other wxWidgets users will chime in on Win32 Dark Mode, so we can share our journey.

That's our current status. We will likely invest a significant effort into investigating the ongoing Win32 Dark Mode effort of the Notepad++ project. We don't know, whether our effort will bring fruits or not as we have no source code example of any successful Win32 application supporting the Dark Mode.

MikeRatcliffe, 2021-04-07:
@bubnikv That has all changed recently...

Dark mode was only fully implemented for UWP apps because there were plans to use WinUI to allow C++ Win32 apps to use UWP controls... this work has now been done.

Because you can now host UWP controls inside your C++ Win32 app these controls will automatically apply dark mode settings when the setting is changed ;)

To do this in a C++ Win32 app you need to use XAML islands.

There are instructions and examples here:

https://docs.microsoft.com/en-us/windows/apps/desktop/modernize/using-the-xaml-hosting-api

I would implement it myself but I am way too busy to contribute at the moment.

bubnikv, 2021-04-08:
@MikeRatcliffe 

Frankly I do not understand your response. We wrote our application in wxWidgets, which pack Win32 controls. Do you propose to turn PrusaSlicer into an UWP application, which would embed an XML island, which will then embed the whole PrusaSlicer Win32 application? Would then the UWP frame take care of skinning the Win32 controls inside the XML island?

We know there is https://github.com/microsoft/ProjectReunion , but we don't know whether and how that could help us.

bubnikv, 2021-04-08:
@MikeRatcliffe I believe the XML islands would not help us as we would have to rewrite the complete PrusaSlicer into UWP controls to support the dark mode.

MikeRatcliffe, 2021-04-08:
> Do you propose to turn PrusaSlicer into an UWP application, which would embed an XML island, which will then embed the whole PrusaSlicer Win32 application?

No, read my response below.

> Would then the UWP frame take care of skinning the Win32 controls inside the XML island?

Yes

> @MikeRatcliffe I believe the XML islands would not help us as we would have to rewrite the complete PrusaSlicer into UWP controls to support the dark mode.

@bubnikv The whole of PrusaSlicer would not need to be turned into a UWP application... you can do it one control at a time. That is the beauty of XAML islands. You can change your C++ Win32 app controls to UDP controls without converting your whole app to a UDP app. Then, of course, they would automatically support dark mode.

It's a long way from ideal but it is what it is. I guess it is a lot of work for what most would see as a cosmetic change (there are a whole host of accessibility reasons this should be done as the app is unusable for many people).

Part of Microsoft's motive in creating this is because Edge is Chromium based and they needed it so that Edge can support native dark mode.

Microsoft's Project Reunion is basically a bunch of APIs to allow apps to use Windows features from Win32, WPF, WinForms, and UWP apps. Part of this project is WinUI 3, which is a user interface layer that contains modern controls. Access to WinUI 3 is provided to C++ Win32 apps via the [XAML Islands feature](https://docs.microsoft.com/en-us/windows/apps/desktop/modernize/using-the-xaml-hosting-api).

bubnikv, 2021-04-08:
@MikeRatcliffe Thanks for your effort explaining me the current state of Windows controls and excuse my ignorance, but I am still not sure I understand what you advise.

PrusaSlicer consists of a bunch of wxWidgets/Win32 windows and dialogs. How can we make these windows and dialogs respect the dark mode palette? How can we convince Win32 controls to switch to dark mode? Could we do it simply?

1)
> > Would then the UWP frame take care of skinning the Win32 controls inside the XML island?
>
> Yes

but then 

2)
> The whole of PrusaSlicer would not need to be turned into a UWP application... you can do it one control at a time. That is the beauty of XAML islands. You can change your C++ Win32 app controls to [...]

So which is true? 1) or 2)? Will the skinning apply to the UDP controls only, so we would have to rewrite the controls one by one, or will the skinning apply to the Win32 controls, which are children of some UDP frame?

What would be the simplest (least effort) technical approach to switch the existing Win32 controls into a dark mode?

Maybe we could have a quick call, let's say on Skype?

CaLviNx, 2021-04-11:
I have a question: if it is so difficult how has "ideamaker" managed to implement dark mode then?

YuSanka, 2021-04-16:
OK, guys. We have a draft implementation of the DarkMode for Window. For us it looks quite ugly, but it's what we have for "quick hand" without putting the hand to wxWidgets.

You can try it out by replacing the PrusaSlicer.DLL from [here](https://drive.google.com/drive/folders/1mmSL05592j6FbYo5dCae1fWLg2c4S2ny?usp=sharing) in your project folder :

Let us know what you think.

Here are different color palettes :
[image]

[image]

[image]

bubnikv, 2021-04-16:
We also looked around the Microsoft's own applications that are distributed with Windows 10.

The only Win32 based application that supports the dark mode is File Explorer. Thus Microsoft only hacked Win32 to support File Explorer. The File Open dialogs that most of Windows applications open are provided by File Explorer, thus they are skinned to dark mode as well.

Notepad, Regedit, Devmgr, sysinternal tools and other Win32 applications from Microsoft do NOT support dark mode.  I have to say this army of Microsoft engineers is a bunch of lazy folks, at least I have to drive this conclusion from some of the comments in this thread. 

C# WPF applications as Visual Studio support dark mode, the "new" Windows 10 applications using the new UI API as the new Calculator or the new Settings application naturally support the dark mode as well. I don't think Microsoft will ever implement Win32 dark mode, more likely they will port the rest of their applications into their new UI framework. They have the money and an army of engineers to do that.

Cura and Ideamaker are written using the Qt framework. Neither of these applications implement dark mode for dialogs, they only implement some dark mode skinning for the main application window. Cura does not skin the menu bar, Ideamaker just skins the menu bar, not the menus. It is a mess and I do not blame them.

Now the question is: How much time should Prusa Research developers spend on tweaking Win32 API to look somehow OK? Is what we have achieved passable or too ugly? Would you use it? How do you like the dark mode support in Cura or Ideamaker?

Personally I would rather spend time working on something more meaningful, for example the tree supports.

jokohoko, 2021-04-16:
I would definitely like the first color pallet most. But would like the titles on the right to also be darker. They stand out to much at the moment.
The same for the support selection box that is white instead of dark like all the other selection boxes. 
Ideally in a dark mode you should never have a background or element with a light fill color. Only text or borders in a lighter style.

bubnikv, 2021-04-16:
It surprised me, that the copy progress dialog of Windows 10 is not dark moded either.
[image]

coopermoto, 2021-04-16:
Yes please. Dark mode all the way.

jgeater, 2021-04-16:
yes, I love Dark mode in my apps.

bubnikv, 2021-04-16:
@coopermoto @jgeater yes what?

We really need a fine grained feedback. None of us developers working on Windows likes the dark theme, thus we don't know what  is passable and what is not and why you love it. Is it because it is easy on your eyes? Could we release one of the builds we posted? Which one? What is the worst eye sore in it in regard to the dark mode palette?

jgeater, 2021-04-16:
I use Dark mode in every Windows app that supports, mostly Office, Visual studio and visual studio code. It is far easier on my eyes and in many cases the contrast makes it easier to see warnings and errors. Specifically the yellow and red colors used with misspellings, errors, and warnings stand out better for me on a dark background.

Does that help?

Ferferite, 2021-04-16:
Replaced the dll file (using Dark with no border buttons), but I keep getting an error: "Unhandled unknown exception; terminating the application", if I click ok(the only button) I get to another error which says:

> can't open file 'D:\Programs\Prusa3D\PrusaSlicer\resources\icons\equal.png' (error 2: The system cannot find the file > specified.)
>  Failed to load image from file "D:\Programs\Prusa3D\PrusaSlicer\resources\icons\equal.png".
>  Internal error: Could not load bitmap: equal

If I replace the dll file with the old one, it works just fine.

YuSanka, 2021-04-16:
@Ferferite, Yes, you are right. In official released build missed some icons. Ok, I will put one of our last build to the link.

YuSanka, 2021-04-16:
[Here](https://drive.google.com/file/d/1bFA2MeYbjoDJ78K7Fvj3uRgaXHaYAESs/view?usp=sharing) is a build?

Ferferite, 2021-04-16:
Works perfectly now, thank you

MikeRatcliffe, 2021-04-16:
@bubnikv I love all three but I like [the third one](https://user-images.githubusercontent.com/19184891/115006862-a5028700-9ea9-11eb-9b09-c5bea49a2bad.jpg) best... it just "feels" better.

I wouldn't change anything from that screenshot.

Sorry, I didn't notice your comment asking to Skype. I don't have any experience with wxWidgets and the library is a pain to work with on Windows so I couldn't create an example.

It looks like you have found a way to implement it anyhow. I can understand your devs not liking dark themes if they are not used to working with them. I used to hate them until we implemented dark mode in Firefox DevTools and for me it was truly life changing.

Personally, I use a dark theme on almost everything because looking at bright screens triggers serious migraines for me (much more serious than a headache).

To be honest I don't think many people care whether it faithfully follows the Windows dark theme.

An easy :thumbsup: from me!

johny-mnemonic, 2021-04-16:
> [Here](https://drive.google.com/file/d/1bFA2MeYbjoDJ78K7Fvj3uRgaXHaYAESs/view?usp=sharing) is a build?

How to switch this build to dark theme? I have Win 10 with selected dark theme, but this is still starting in light theme same as standard 2.3.0 build.

I also tried downloading dll/exe files and replacing them, but that produces lot of errors about missing dll files (mostly their name ends with "D" so I guess those are debugging libs?).

As other guys mentioned, any dark theme is better then default light theme. The reason is, that especially when you have system and all other apps in dark theme, then having single window in light theme is tiresome for the eyes and prolonged use hurts.
It is really interesting, that none of you uses dark theme :-O I work in the company of 150+ developers and more than 60% use dark theme I would say.

DesC21, 2021-04-17:
This is awesome progress, thanks so much for doing this. As the others said having a dark theme is so much easier on the eyes for me. Windows is really lazy with getting this to work on everything... but thats another thing...

I like Version 3 best (Dark with no border buttons). It is a great step and good to use as is it if you really are a fan of dark themes but there are some things which would be nice if they are "fixed" if you roll it out in 2.4.0 because people might think of it as unfinished/buggy:

1.  When you switch tabs, the initial highlighted element is white which looks super odd and you can't really read the text. When you click on the elements it switches to the "correct" highlight color which is blue. I don't know if this is easy fixable (since the highlight color is already correct). Maybe the highlight color is only changed after the first click (not sure how you do it, I never looked at the code)? 
[image]

2. the spacing on the tabs looks super weird now since there is no clear distinction between the elements. On the light theme you can see the "tabs" so it looks fine. Maybe you can change that with different colors or borders to make it look better? 
[image]

3.  When you click on a part in the plater view, you can't read the part name in the table because the highlight color is white. If you click on the part in the table itself it works fine.
[image]

4. the "info field" still has the black font color which is super hard to read (Plater tab -> add .stl -> and click on the part). change to white should fix this "easily" :-)
[image]

Ok and additionally something which is not really necessary but it would make the dark theme pretty much perfect I think.... The drop down menus and text fields are "bothering" me as a dark theme user. Most of them are still super light gray which kinda defeats the sense of the dark theme because you have the light colors jumping at you. But some of them are dark and super nice. I don't know if there is a reason for this or if you simply didn't bother to switch all elements yet. On the Plater view for example you switched the "Print settings, Filament, Printer" to dark but the Supports "drop down" is still grey (at least the choosen item, the list itself is dark already). If you switch that to the same style as the three above it, it will look a lot better.
[image]

Same goes for the elements in the setting tabs. If you could make them the same color as the "Print settings, Filament, Printer" Drop Down menus it will look sooooo much better. 
[image]

And the last thing (promise :-)), if it is possible I would prefer to have the border color of the elements not white but a slightly lighter gray than the background. but this is really nit picking ;-). This is an example on how it looks on linux for me just as an idea on how the colors work together. 
[image]

Thank you so much for working on this. I do understand that this is simply nice to have and not essential feature but I really think quite a few people would like this. If you need any more feedback, let me know I am more than happy to help.

DesC21, 2021-04-17:
> How to switch this build to dark theme? I have Win 10 with selected dark theme, but this is still starting in light theme same as standard 2.3.0 build.
>
> I also tried downloading dll/exe files and [...]

You need to run the Prusaslicer 2.4.0 alpha to make this work. Goto the original link that was posted here. Then click on download all (right top corner). 

> OK, guys. We have a draft implementation of the DarkMode for Window. For us it looks quite ugly, but it's what we have for "quick hand" without putting the hand to wxWidgets.
>
> You can try it out by [...]

After download unzip the file and then you get the new PrusaSlicer alpha folder and the three dark theme variants. If you start the alpha as is it starts in light mode but you can now copy the dark themes files over to the alpha folder and check the new themes. 

PS.: I know I am going to test the fuzzy walls feature this weekend :-)

johny-mnemonic, 2021-04-17:
@DesC21 As I described, I did exactly what you are suggesting. 
I downloaded the 2.4.0 alpha build, but it starts in light mode (even though the "welcome screen" is dark). And when I tried to copy over any of the exe/dll files in the dark theme folders, it doesn't start at all, complaining about missing dll files like e.g. 
`ucrtbased.dll`, `VCRUNTIME140D.dll` or `MSVCP140D`. Notice the letter "D" at the end of the filename. That's why I wrote, it looks like it is missing debugging variants of the regular libs.

YuSanka, 2021-04-18:
@johny-mnemonic , you have to raplace just PruSlacer.DLL

fernandex00, 2021-04-18:
replace the PrusaSlicer.dll from darkmode folder to PrusaSlicer.dll from pusaslicer 2.4 and the app wont start, it working again if replace with the orginal PrusaSlicer.dll

johny-mnemonic, 2021-04-18:
@YuSanka I can confirm the same behavior as @fernandex00 describes. With only dll replaced, exe does nothing. No errors just short spinning wheel and then nothing happens.

bubnikv, 2021-04-19:
@YuSanka provided a DLL, which was compiled against debugging variants of Visual Studio libraries. She will post the release DLL soon.

bubnikv, 2021-04-19:
@DesC21
> Windows is really lazy with getting this to work on everything.

I used the word "lazy" with a tongue in cheek. They likely have their technical reasons. Windows is the only platform that really allows you to run very old executables. Microsoft invests huge effort in maintaining backwards compatibility, which is both their blessing and their curse.

YuSanka, 2021-04-19:
Ok, just try this [build ](https://drive.google.com/file/d/1Skn7qiUJ1JxB7pKAMNdz7JgAw5WLcBdg/view?usp=sharing)without any replacing :) It's a build of third variant od DarkMode

DesC21, 2021-04-19:
> @DesC21
>
>
> I used the word "lazy" with a tongue in cheek. They likely have their technical reasons. Windows is the only platform that really allows you to run very old executables. Microsoft invests [...]

me too :-), it wasn't meant to be offensive. I am sure they have their reasons and a lot of other more important topics to work on but since they announced the dark mode stuff not much happened to be honest (as is seen with the various programs that are not supported even from microsoft themselves). Anyway, I like that you guys invested some work to make this work!

fernandex00, 2021-04-19:
It's working now, looks good, the thing I notice is that the dark theme works following the system theme, if I'm using the default color, prusaslicer will be using the default theme, will be nice to have the option to always have the app themed

matthewlloyd, 2021-05-12:
Dark mode is the main thing keeping me from switching from Cura, which has an excellent dark color scheme.

> Ok, just try this [build ](https://drive.google.com/file/d/1Skn7qiUJ1JxB7pKAMNdz7JgAw5WLcBdg/view?usp=sharing)without any replacing :) It's a build of third variant od DarkMode

Thanks for uploading this, I just gave it a try. Here is some feedback. There are some minor visual issues:

- The black text in the status bar is unreadable against the dark grey background.
- Buttons and tabs which normally are outlined (e.g. 'Simple, Expert, Advanced' on the right hand side, or the tabs at the top) no longer have outlines.
- Some controls have the wrong width or height, e.g. the tabs, and drop-down selects (e.g. Infill on the right).
- The menu bar is white text on a black background, which is at odds with other programs supporting a dark theme.
- The background of the 3D view is still very bright in dark mode, and there's no way to change it.

Looking forward to the next version.

matthewlloyd, 2021-05-13:
For those looking for a "quick fix", NegativeScreen works quite well and works with any version of PrusaSlicer (or any other app you like). It's not perfect but it is very easy to turn on and off with a keyboard shortcut.

https://zerowidthjoiner.net/negativescreen

bubnikv, 2021-08-11:
Based on the popular request, we have implemented Dark Mode for Windows and we have also fixed the High Contrast mode #3038.

While PrusaSlicer supports dark mode on OSX and Linux for some time already, Dark Mode on Windows was the one from the hard basket. PrusaSlicer's user interface is implemented using a [wxWidgets](https://www.wxwidgets.org/) multi-platform library, which wraps around the native controls the platform provides. On Windows, wxWidgets wrap around the aging Win32 controls, for which there is no official Dark Mode support provided by Microsoft since the introduction of the Dark Mode into Windows 10 in fall 2018. Unfortunately Microsoft just implemented an undocumented API for the controls used by Windows Explorer to switch to Dark Mode. If a Win32 control is not used by Windows Explorer main window, it is not Dark Moded and it never will be, and the API for those controls that are supported is not official, thus it may be changed without notice.

We have done our best using the undocumented Dark Mode API and skinning of Win32 controls. We even implemented our own replacements for some Win32 controls, which were not skinnable, for example the tabbed notebook that we use to switch between the Plater / Print / Filament / Printer profiles (this one fixes speed of switching the tabs on 4k displays #6309). Still some of the controls are not skinned properly, for example the scroll bars, status bar or the spin control buttons, thus PrusaSlicer may not look quite polished in Dark Mode. Therefore for now we decided to not switch PrusaSlicer into Dark Mode following the Windows system settings, but we are letting the user to select dark mode from Preferences using the "Use Dark color mode (experimental)" checkbox.

As an alternative to the "tabbed notebook" control to switch between the Plater / Print / Filament / Printer profiles, we newly allow these tabs to be placed into the application menu, which may be welcome on small laptop screens.

Thhe dark mode will be released with PrusaSLicer 2.4.0-alpha1.

closed 2021-08-11 by bubnikv