prusa3d/PrusaSlicer #4494: Feature request: Split view in slicer

Sundheden, 2020-07-07:
I would love to have a "split view", somewhat similar to the workflow in Cura, where you immediately can see the changes you make to the profile, being apply to the sliced model as I change them. Reference: [https://i.imgur.com/Ftj7RXu.png](https://i.imgur.com/Ftj7RXu.png)

n8bot, 2020-07-07:
This is basically coming. 

In the alpha version 2.3.0, the print/filament/printer settings can be popped out into their own window, which you could put side by side with the main window if you’d like.

MadeforMaking, 2020-10-19:
Can this be found in the current 2.3.0-alpha1? I don't think so but enquiring.

n8bot, 2020-10-19:
Yes. Enable the options marked in red below:
[image]

MadeforMaking, 2020-10-28:
Thanks this is neat. 👍🏻

rtyr, 2021-02-22:
Implemented in 2.3.0. Closing.

closed 2021-02-22 by rtyr