prusa3d/PrusaSlicer #4495: Severe Overextrusion of last (filling) path at end of print

dartrax, 2020-07-07:
### Version
2.2.0 win 64 and 2.3.0-alpha0 (13a2e8e)

### Operating system type + version
PrusaSlicer
Version:   2.2.0+win64
Build:     PrusaSlicer-2.2.0+win64-202003211145

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19041), 64-bit edition

### Behavior
Severe Overextrusion of the last filling path at end of print. I first thought it was a rendering issue but E is going up in GCode. Somehow related to the orientation of the model, if I rotate around Z it disappears.
[image]

#### Project File (.3MF) where problem occurs
[Winkeladapter.zip](https://github.com/prusa3d/PrusaSlicer/files/4885942/Winkeladapter.zip)

lukasmatena, 2020-07-09:
It looks quite interesting, but I cannot reproduce it. The 3MF slices fine on my computer.

> E is going up in GCode

You are using absolute addressing for the extruder, so the sole fact that E is increasing is fine. But you are correct that it progressively extrudes more as x increases (judging by our gcode). The gap fill line seems to be broken into many pieces (in a fashion similar to e.g. #4106) and I guess the absolute E somehow does not work then. Maybe.

We will try to reproduce it. In the meantime, could you check if it disappeares with relative extruder addressing (Printer Settings->General->Use relative E distances)? Thanks.

dartrax, 2020-07-09:
Hi,
that was at work, I've tried to reproduce at home with 2.3.0-alpha0 (13a2e8e) and the .3mf-file, results are the same - should be reproducable.

I've tried with relative E distances - preview looks the same. GCode is here:
[Winkeladapter_relative-E.zip](https://github.com/prusa3d/PrusaSlicer/files/4897732/Winkeladapter_relative-E.zip)

It goes away when I disable thin_walls (of course) or rotate only a little bit. However, rotating exactly 90° (in either direction) also triggers the issue. But not when rotating 180°.

FidelCapo, 2020-07-10:
I can reproduce this issue on win. It goes away if I scale object to 100.2% too.

closed 2026-09-29 by prusaslicer-bot[bot]