prusa3d/PrusaSlicer #4496: Model below bed after slice for custom printer config -UP BOX+

kstith68, 2020-07-07:
### Version
2.2.0+Win64

### Operating system type + version
Windows 10 Home x64
10.0.18364 build 18362

### 3D printer brand / version + firmware version (if known)
Tiertime UP BOX+

### Behavior
I had to use a weird configuration for the X, Y, and Z to get the G-code on my UP BOX+ printer.  If fact the Z offset is -209.05.  I think this is the root of the problem but the PrusaSlicer generated G-code works in my printer.  I can add a model and it looks fine on the platform in the Platter tab but once I slice the model in is displayed below the platform.  You can't zoom the sliced model since the platform is fixed in the middle of the screen and the sliced model is off the screen when you zoom. 

[image]
[image]
[image]

[PrusaSlicer_config_bundle.zip](https://github.com/prusa3d/PrusaSlicer/files/4886784/PrusaSlicer_config_bundle.zip)

_Is this a new feature request?_
I'm not sure if this is a bug or a new feature request.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2020-07-07:
Z offset option from upstream Slic3r was meant for different purpose. It can be used to slightly compensate for an incorrectly calibrated Z end-stop. Print preview just reflects your Z offset value (and generated g-code). It is not expected, that initial height is at Z-210mm. It seems, that your printer prints from negative MAX to zero by default.

I believe there is a possible solution. You can try to add `M206 Z-209.05` command to your start g-code and set z offset in PrusaSlicer back to 0.

[image]
[image]

kstith68, 2020-07-07:
Hi ytyr,
I made the changes you recommended, sliced a model and saved the G-code.  When I looked at the G-code I noticed that once it starts printing it issues a "G1 Z0.150 F7800.000" command.  I think that would crash the printer.  Do you agree?  I attached the G-code.

[xyzCalibration_cube.gcode 25scale.zip](https://github.com/prusa3d/PrusaSlicer/files/4887598/xyzCalibration_cube.gcode.25scale.zip)

rtyr, 2020-07-08:
It should't crash, M206 sets offset on the printer side so it should work, I saw someone recommending this workaround on UP forums (for different model). I can't garantee it will work fine for your printer of course.

kstith68, 2020-07-08:
Success! I printed the generated G-code and it does work.  Thank you for your assistance.

**Much appreciated!!**

closed 2020-07-08 by kstith68

Ljun-1994, 2023-06-14:
Hello, have you continued to improve Prusa control of UP BOX+? Can you ask for advice?
I want to use Prusa to control UP BOX+now. Do you have any UP BOX+accessory files here。

kstith68, 2023-06-14:
Unfortunately, I haven't used PrusaSlicer with the UP Box in years.  Sorry.
Ken

Ljun-1994, 2023-06-15:
Thank you
Can you provide me with the parameter configuration of your Prus UP BOX+for reference? I am currently conducting experiments and would like to see other references

kstith68, 2023-06-15:
What parameters are you looking for? I use UPStudio and it's rather closed when it comes to settings.

Ljun-1994, 2023-06-15:
[image]

You successfully connected to BOX using Prusa and printed it successfully. Do you still have Prusa setup parameters?

I was testing Prusa program last night, and the printer crashed. It should have been a height issue, and I needed to make a part that I haven't been studying yet. Today, I am researching.

Are you using UP Studio 3? I also want to delve deeper into this software. Are there any related discussion posts or forums?

kstith68, 2023-06-16:
I have not used PrusaSlicer on this computer. I used a program to move my programs and data files from my old PC to the new PC, so I at least have these settings shown below.  My version of UPStudio is 2.6.71.627.

Ljun-1994, 2023-06-16:
I can't see the settings shown below
May I ask if there are many people using UP Studio 3? Do you use forums or deep parsing like Prusa does
[image]

kstith68, 2023-06-17:
I haven't upgraded and I don't know about others as I'm not active in the forums.