prusa3d/PrusaSlicer #4499: Error gcode_flavor

manurafal, 2020-07-08:
### Version
2.2.0

### Operating system type + version
Windows 10 Pro

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro

### Behavior

I put this code in Filament Gcode but it does not work.

{if gcode_flavor=="marlin"}; Filament gcode
M900 K30; Filament gcode
{endif}; Filament gcode

### Message Error
The error is: "G-code export to C:\Users\XXXX\AppData\Local\Temp\.3484.gcode failed due to invalid custom G-code sections:
start_filament_gcode
Please inspect the file C:\Users\XXXX\AppData\Local\Temp\.3484.gcode.tmp for error messages enclosed between
!!!!! Failed to process the custom G-code template ...
and
!!!!! End of an error report for the custom G-code template ..."

n8bot, 2020-07-11:
Can you do as the message says, then post the results here?

>Please inspect the file C:\Users\XXXX\AppData\Local\Temp.3484.gcode.tmp for error messages enclosed between
!!!!! Failed to process the custom G-code template ...
and
!!!!! End of an error report for the custom G-code template ..."

manurafal, 2020-07-13:
This is the content of the file:

!!!!! Failed to process the custom G-code template start_filament_gcode
Parsing error at line 2: Unknown scalar variable type
{if gcode_flavor=="marlin"} ; Si tipo de codigo G es igual a Marlin
    ^
!!!!! End of an error report for the custom G-code template start_filament_gcode

manurafal, 2020-07-13:
> 
> 
> Can you do as the message says, then post the results here?
> 
> > Please inspect the file C:\Users\XXXX\AppData\Local\Temp.3484.gcode.tmp for error messages enclosed between
> > !!!!! Failed to process the custom G-code template ...
> > and
> > !!!!! End of an error report for the custom G-code template ..."

This is the content of the file:

!!!!! Failed to process the custom G-code template start_filament_gcode
Parsing error at line 2: Unknown scalar variable type
{if gcode_flavor=="marlin"} ; Si tipo de codigo G es igual a Marlin
^
!!!!! End of an error report for the custom G-code template start_filament_gcode

n8bot, 2020-07-13:
Could it be that 'marlin' needs to be capitalised? "Marlin"

manurafal, 2020-07-13:
> 
> 
> Could it be that 'marlin' needs to be capitalised? "Marlin"

It´s a string. This has to compare a variable with a string.

n8bot, 2020-07-13:
Yes, but the value listed in the options in software is capitalised, 'Marlin,' so it could be that only that exact string will match.

manurafal, 2020-07-21:
How can I make the following statement?

{if gcode_flavor=="Marlin"}
xxxxxxxxx
xxxxxxxxx
{endif}

I don't know what I can do if I can use the variable gcode_flavor.

closed 2020-07-21 by manurafal