prusa3d/PrusaSlicer #4501: [Feature Request] Align models on bed when manually placing them

TraylorHead, 2020-07-08:
### Version 2.2.0

Asking for a alignment feature so when placing many objects you can select several, then have them alight (Left, Top, Right, Bottom) to last one selected.

You could also add a spacing to gap them an equal distance between the closer and farthest.  If over lap occurs it's up to user to re-space and try again.

Thanks

BillyCroan, 2022-09-16:
I proposed similar on the forums: https://forum.prusa3d.com/forum/prusaslicer/feature-request-professional-alignment-tools/#post-620469 with perhaps a little more verbosity:

Very often I find my self plating up a dozen unique objects for printing.  I don't like using Auto-arrange for a couple reasons:

    It looks random.  Which might be okay if it optimizes print time, but there's other problems...
    It's less efficient.  I can pack things in 50-60% tighter if I think about what goes where
    It can put rectangular objects diagonal which causes dual axis movement that could just be single
    It doesn't consider which side of the objects it best to face the edge of the build plate.  I think if one side overhangs for example, that side should face the plate edge so that the colder edge of the plate isn't in contact with as much object.
    It won't nest objects.  Very often I have objects with large hollows that could nest other objects easily to maximize production of a single print.  I think auto-arrange treats each object like a rectangle of its maximum possible dimensions.
    I always have items "off to the side" that I plan to print on another day or maybe already did but haven't verified they came out all right.  I use the location of each object (on the plate or not) as a boolean for "do I want to print this or not"

I start my own jobs but I'm not the person who takes them off the printer.  I'd like that person to arrive to a nice looking plate that doesn't look like I was on a moving boat when I was laying things out.  And I don't want to spend hours reading and typing specific offset numbers in by hand.

I propose a couple tools to help us professionally prepare multiple objects for printing:

    Highlighting/alerts when objects are "too close" to each-other (buffer distance select-able by user)
    Enable/disable-able snap to grid
    Snap to edge or centerline of plate
    Snap to edge or centerline of other object (with buffer distance)
    Select group of objects and "align x/y/z axis"
    Select group of objects and "space evenly along x/y/z axis

#'s 2-6 are basically things that Google Draw and various word processors do, which seems an odd thing to emulate here, but i think this sort of tidyness can make better use of the build plate, and elevate the perception of 3d printing that much more.

fscheps, 2024-05-21:
Is the alignment functionality coming? I found today that I needed to do a whole in a piece and I wanted to center it to the object but I couldn't find any way to do this, so the only way of doing so if bringing into an editor to do this and then slice it in Prusa Slicer...thank you!

Tupson444, 2025-06-01:
Please add this functionality (nothing complex like align faces, just align position on the plate: horizontal/vertical, left/right, etc.). I'm placing many small pieces on the plate, and I have to type coordinates for each one to align them (auto arrange is not an option for me in this case). It would be much simpler if I could select a whole row of instances and click "align horizontally (center)".

Jan-Soustruznik, 2025-08-08:
close as duplicate of #1425

closed 2025-08-08 by Jan-Soustruznik

Tupson444, 2025-08-08:
This is not a duplicate of #1425, it is different. This issue suggests aligning many objects on the build plate (like the arrange tool, but with more manual control), not joining two objects. Please reopen.

JanTuts, 2025-09-06:
As @Tupson444 said, this feature request is not at all about snapping parts together, to join them, like #1425.
This one is about relative alignment and spacing of multiple parts to each other, and/or the print bed, [like for example the _Align_ panel in Adobe Illustrator](https://youtu.be/M42mQ5Nd6wI?feature=shared&t=28). 
So indeed, please reopen.