prusa3d/PrusaSlicer #4503: 3MF import does not respect basematerial and related tags and attributes

tg73, 2020-07-09:
### Version
2.2.0+win64

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
_Actual Behaviour:_
File -> OpenProject, select a 3mf (not generated by PS) which uses the `basematerial` and related tags and attributes according to the [3MF specification](https://github.com/3MFConsortium/spec_core/blob/master/3MF%20Core%20Specification.md#chapter-5-material-resources), with materials applied per-object. PS does not appear to make any attempt to use the material information, and assigns extruders in a repeating cycle.

[image]

I also note that from casual inspection, `src/libslic3r/Format/3mf.cpp` does not appear to be aware of the `basematerial` or `base` tags, or the `pid` or `pindex` attributes, which might explain the behaviour.

_Expected Behaviour:_
In the simple case where the 3MF contains a single `basematerial` element, and material is assigned per-object (rather than per-triangle or per-vertex), PS should map `pindex` to extruder number. For objects with `pindex` greater than the number of extruders configured, PS should either assign the default extruder, or pindex modulo the number of available extruders, to be discussed.

There is clearly scope for smarter support, such as mapping extruders based well-known strings in `base/@name` to material types (eg, "PLA"), and further refining the selection by finding the nearest matching colour. However, personally I'm not looking for that kind of functionality.

#### Project File (.3MF) where problem occurs
[test11.zip](https://github.com/prusa3d/PrusaSlicer/files/4897636/test11.zip)

bubnikv, 2021-09-22:
How did you create such a 3MF?

> In the simple case where the 3MF contains a single basematerial element, and material is assigned per-object (rather than per-triangle or per-vertex), PS should map pindex to extruder number. For [...]

I don't think that mapping any 3MF property to FDM extruder ID is specified by a 3MF specification.

tg73, 2021-09-24:
The 3MF was generated by my own software, which algorithmically generates plates of multi-material objects. I used `basematerial` et al as my first attempt to generate a multi-material 3MF for consumption by PS. Those properties seem to be the obvious (only?) choice for expressing different materials in a model without relying upon custom extensions to 3MF. The mapping of material to extruder ID was suggested as a reasonable default behaviour for PS, allowing it to recognise and respect in some useful way a standards-based expression of different materials in a 3MF.

Since this issue was filed, I now generate the custom, non-standard configuration and metadata that PS itself emits. This goes as far as emitting seam enforcement metadata. As per #5294, I do wish that PS could work with 3MF more in the spirit of XML and the transparency and interoperability aims of 3MF, but I also recognise the performance concerns when it comes to encoding and parsing painting metadata. Generally, PS seems to rely heavily on custom metadata and configuration, often eschewing 3MF standards-based approaches even when available: this really does not help with portability/interoperability. If I remember correctly, pretty much the only data that's portable are the volumes, but even the tree of volumes is encoded in an unusual way which is more PS-native than 3MF-native. It's a bit "3MF in name only".

bubnikv, 2021-09-24:
: If I remember correctly, pretty much the only data that's portable are the volumes, but even the tree of volumes is encoded in an unusual way which is more PS-native than 3MF-native. It's a bit "3MF in name only".

This is unfortunate implementation, which we will fix one day. We plan to revamp our 3MF format, however we plan to do it in a single shot, as these new 3MFs will likely not be consumable by older PrusaSlicer versions.

SupItsSaige, 2024-05-16:
PLEASE PLEASE! This is so needed!!!! Especially with the Prusa XL finally making its way out, how is there not a more streamlined process for bringing in multi-color/material models? I get that a lot of people have been use to needing to manually separate the colors through the mesh itself as separate parts, but that is not efficient and definitely not possible for some of the more complex models out there.

Being able to keep the mesh color metadata from the 3D-modeling software would make the process so much more efficient and tuning a model would be so much easier. Is there any update at all for this? It is pretty crazy that what was such a useful request 4 years ago still has not been implemented. This is absolutely a must have for those of us that are using Prusa XLs and MMUs for business purposes and have a lot of iterations.... Otherwise we are just making the process unnecessarily more difficult by keeping things status quo (manually breaking apart the meshes).

SupItsSaige, 2026-09-20:
PLEASE DON'T CLOSE! This is still a much needed tool especially with how Prusa is working on more multi color printers and implementing ColorMix. This is needed more than ever and this ticket has been open more than SIX YEARS! With people still wanting this type of functionality, I really hope it can finally get implemented to allow for much better work flows for PrusaSlicer users. Come on guys even Bambu allows for color importing!