prusa3d/PrusaSlicer #4504: PrusaSlicer reporting "No layers were detected" in command line mode

SChmElmos, 2020-07-09:
### Version
Version 2.2.0+win64

### Operating system type + version
Windows 10 64bit

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3

### Behavior
When using Prusa Slicer from the command line to automatially export gcode, it returns the following Message:
No layers were detected. You might want to repair your STL file(s) or check their size or thickness and retry.

Loading the same Model in GUI Mode produces no errors at all.
Tried with fresh data dirs and different config.ini's via --load parameter (also attached)

Meshlabs shows the file just fine, too (no inverse vertices etc)
The same is reproduceable with a set of 3 of the tried 5 files, all three work just fine when sliced in gui mode.

Steps to reproduce:
Call Prusa Slicer with attached stl file
./prusa-slicer.exe -s -g -output C:\\Temp\\slicer\\test\\2ee93f63-61e6-4757-ad41-cbebd106d098.gcode C:\\Temp\\slicer\\test\\2ee93f63-61e6-4757-ad41-cbebd106d098.stl --loglevel 5

Expected Result:
Slicer exports a gcode file to the given location

Prusa Slicer Log:
$ ./prusa-slicer.exe -s --load C:\\Temp\\slicer\\test\\slicer.ini -g -output C:\\Temp\\slicer\\test\\2ee93f63-61e6-4757-ad41-cbebd106d098.gcode C:\\Temp\\slicer\\test\\2ee93f63-61e6-4757-ad41-cbebd106d098.stl --loglevel 5                                                   No layers were detected. You might want to repair your STL file(s) or check their size or thickness and retry.                                                                                                                                                                                                                                                                                                                                                                                                                                                  [2020-07-09 17:02:36.283901] [0x00004170] [debug]   TriangleMesh::repair() started                                                                                                                                                                                              [2020-07-09 17:02:36.288890] [0x00004170] [debug]   TriangleMesh::repair() finished                                                                                                                                                                                             [2020-07-09 17:02:36.288890] [0x00004170] [trace]   TriangleMeshSlicer::require_shared_vertices - start                                                                                                                                                                         [2020-07-09 17:02:36.288890] [0x00004170] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices                                                                                                                                                  [2020-07-09 17:02:36.288890] [0x00004170] [trace]   TriangleMeshSlicer::require_shared_vertices - end                                                                                                                                                                           [2020-07-09 17:02:36.308806] [0x00004170] [debug]   TriangleMesh::repair() started                                                                                                                                                                                              [2020-07-09 17:02:36.310800] [0x00004170] [debug]   TriangleMesh::repair() finished                                                                                                                                                                                             [2020-07-09 17:02:36.310800] [0x00004170] [trace]   TriangleMeshSlicer::require_shared_vertices - start                                                                                                                                                                         [2020-07-09 17:02:36.310800] [0x00004170] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices                                                                                                                                                  [2020-07-09 17:02:36.310800] [0x00004170] [trace]   TriangleMeshSlicer::require_shared_vertices - end                                                                                                                                                                           [2020-07-09 17:02:36.315787] [0x00004170] [info]    Staring the slicing process. WorkingSet: 32MB; PrivateBytes: 26MB; Pagefile(peak): 26MB(32MB)                                                                                                                               10 => Processing triangulated mesh                                                                                                                                                                                                                                              [2020-07-09 17:02:36.315787] [0x00004170] [info]    Slicing objects... WorkingSet: 32MB; PrivateBytes: 26MB; Pagefile(peak): 26MB(32MB)                                                                                                                                         [2020-07-09 17:02:36.315787] [0x00004170] [debug]   Slicing objects - region 0                                                                                                                                                                                                  [2020-07-09 17:02:36.320774] [0x00004170] [trace]   TriangleMeshSlicer::require_shared_vertices - start                                                                                                                                                                         [2020-07-09 17:02:36.320774] [0x00004170] [trace]   TriangleMeshSlicer::require_shared_vertices - end                                                                                                                                                                           [2020-07-09 17:02:36.323766] [0x00004170] [debug]   TriangleMeshSlicer::slice                                                                                                                                                                                                   [2020-07-09 17:02:36.323766] [0x00004170] [debug]   TriangleMeshSlicer::_slice_do                                                                                                                                                                                               [2020-07-09 17:02:36.324763] [0x00004170] [debug]   TriangleMeshSlicer::_make_loops_do                                                                                                                                                                                          [2020-07-09 17:02:36.324763] [0x00004170] [debug]   TriangleMeshSlicer::slice finished                                                                                                                                                                                          [2020-07-09 17:02:36.324763] [0x00004170] [debug]   TriangleMeshSlicer::make_expolygons in parallel - start                                                                                                                                                                     [2020-07-09 17:02:36.325759] [0x00004170] [debug]   TriangleMeshSlicer::make_expolygons in parallel - end                                                                                                                                                                       [2020-07-09 17:02:36.325759] [0x00004170] [debug]   Slicing objects - append slices 0 start                                                                                                                                                                                     [2020-07-09 17:02:36.325759] [0x00004170] [debug]   Slicing objects - append slices 0 end                                                                                                                                                                                       [2020-07-09 17:02:36.325759] [0x00004170] [debug]   Slicing objects - removing top empty layers     [... cut ...]                                           [2020-07-09 17:02:36.325759] [0x00004170] [debug]   Slicing objects - fixing slicing errors in parallel - begin                                                                                                                                                                 [2020-07-09 17:02:36.325759] [0x00004170] [debug]   Slicing objects - fixing slicing errors in parallel - end

[files.zip](https://github.com/prusa3d/PrusaSlicer/files/4897709/files.zip)

SChmElmos, 2020-07-09:
Log Reupload with correct line endings (better to read)

$ ./prusa-slicer.exe -s --load C:\\Temp\\slicer\\test\\slicer.ini -g -output C:\\Temp\\slicer\\test\\2ee93f63-61e6-4757-ad41-cbebd106d098.gcode C:\\Temp\\slicer\\test\\2ee93f63-61e6-4757-ad41-cbebd106d098.stl --loglevel 5
No layers were detected. You might want to repair your STL file(s) or check their size or thickness and retry.

[2020-07-09 17:02:36.283901] [0x00004170] [debug]   TriangleMesh::repair() started
[2020-07-09 17:02:36.288890] [0x00004170] [debug]   TriangleMesh::repair() finished
[2020-07-09 17:02:36.288890] [0x00004170] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2020-07-09 17:02:36.288890] [0x00004170] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices
[2020-07-09 17:02:36.288890] [0x00004170] [trace]   TriangleMeshSlicer::require_shared_vertices - end
[2020-07-09 17:02:36.308806] [0x00004170] [debug]   TriangleMesh::repair() started
[2020-07-09 17:02:36.310800] [0x00004170] [debug]   TriangleMesh::repair() finished
[2020-07-09 17:02:36.310800] [0x00004170] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2020-07-09 17:02:36.310800] [0x00004170] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices
[2020-07-09 17:02:36.310800] [0x00004170] [trace]   TriangleMeshSlicer::require_shared_vertices - end
[2020-07-09 17:02:36.315787] [0x00004170] [info]    Staring the slicing process. WorkingSet: 32MB; PrivateBytes: 26MB; Pagefile(peak): 26MB(32MB)
10 => Processing triangulated mesh
[2020-07-09 17:02:36.315787] [0x00004170] [info]    Slicing objects... WorkingSet: 32MB; PrivateBytes: 26MB; Pagefile(peak): 26MB(32MB)
[2020-07-09 17:02:36.315787] [0x00004170] [debug]   Slicing objects - region 0
[2020-07-09 17:02:36.320774] [0x00004170] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2020-07-09 17:02:36.320774] [0x00004170] [trace]   TriangleMeshSlicer::require_shared_vertices - end
[2020-07-09 17:02:36.323766] [0x00004170] [debug]   TriangleMeshSlicer::slice
[2020-07-09 17:02:36.323766] [0x00004170] [debug]   TriangleMeshSlicer::_slice_do
[2020-07-09 17:02:36.324763] [0x00004170] [debug]   TriangleMeshSlicer::_make_loops_do
[2020-07-09 17:02:36.324763] [0x00004170] [debug]   TriangleMeshSlicer::slice finished
[2020-07-09 17:02:36.324763] [0x00004170] [debug]   TriangleMeshSlicer::make_expolygons in parallel - start
[2020-07-09 17:02:36.325759] [0x00004170] [debug]   TriangleMeshSlicer::make_expolygons in parallel - end
[2020-07-09 17:02:36.325759] [0x00004170] [debug]   Slicing objects - append slices 0 start
[2020-07-09 17:02:36.325759] [0x00004170] [debug]   Slicing objects - append slices 0 end
[2020-07-09 17:02:36.325759] [0x00004170] [debug]   Slicing objects - removing top empty layers
[2020-07-09 17:02:36.325759] [0x00004170] [debug]   Slicing objects - make_slices in parallel - begin
[2020-07-09 17:02:36.325759] [0x00004170] [debug]   Slicing objects - make_slices in parallel - end
[2020-07-09 17:02:36.325759] [0x00004170] [debug]   Slicing objects - fixing slicing errors in parallel - begin
[2020-07-09 17:02:36.325759] [0x00004170] [debug]   Slicing objects - fixing slicing errors in parallel - end

bubnikv, 2021-09-25:
Bump bump bump.

Maybe your object is levitating?

arifbalik, 2022-09-30:
I face the same issue.

I have two models each has dimensions of 10mmx10mmx0.1mm, they are called MultiLayer1.stl and MultiLayer2.stl. MultiLayer2.stl is indeed "levitating" but on purpose, it is offset by 0.1mm in the Z axis in order to have different printheads print different layers. 

`/prusa-slicer-console.exe -s --dont-arrange -m --center 65,45 -o /composition.gcode /composition.amf --loglevel 5`

When I run the command above I get;

```
[2022-09-30 17:08:22.594905] [0x0001bca0] [debug]   TriangleMesh::repair() started
[2022-09-30 17:08:22.595041] [0x0001bca0] [debug]   TriangleMesh::repair() finished
[2022-09-30 17:08:22.596066] [0x0001bca0] [debug]   TriangleMesh::repair() started
[2022-09-30 17:08:22.597756] [0x0001bca0] [debug]   TriangleMesh::repair() finished
[2022-09-30 17:08:22.598054] [0x0001bca0] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2022-09-30 17:08:22.598054] [0x0001bca0] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices
[2022-09-30 17:08:22.598054] [0x0001bca0] [trace]   TriangleMeshSlicer::require_shared_vertices - end
[2022-09-30 17:08:22.598054] [0x0001bca0] [debug]   TriangleMesh::repair() started
[2022-09-30 17:08:22.599061] [0x0001bca0] [debug]   TriangleMesh::repair() finished
[2022-09-30 17:08:22.599061] [0x0001bca0] [debug]   TriangleMesh::repair() started
[2022-09-30 17:08:22.599061] [0x0001bca0] [debug]   TriangleMesh::repair() finished
[2022-09-30 17:08:22.600062] [0x0001bca0] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2022-09-30 17:08:22.600062] [0x0001bca0] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices
[2022-09-30 17:08:22.600062] [0x0001bca0] [trace]   TriangleMeshSlicer::require_shared_vertices - end
```

I can import STL files to the GUI, select different extruders, and slice without any problem. Also, I can merge and convert multiple STL files to one AMF file and slice them only using the command line.

I either get No layers were detected or the output above when I tweak with the print parameters. I have no idea how to solve this.

closed 2026-09-29 by prusaslicer-bot[bot]