prusa3d/PrusaSlicer #4505: Change tool after start code

MASPPrototipi, 2020-07-09:
PrusaSlicer
Version:   2.2.0+win64
Build:     PrusaSlicer-2.2.0+win64-202003211145

When you write the change tool script you select what printer do with preview extruder and what do with the next one.
The problem is that the change script run after the start code, having a T-1 that cause problems.
You can easly resolve that by devide the script from BEFORE CHANGE AND AFTER CHANGE and after the start code starts only AFTER THE CHANGE

lukasmatena, 2020-07-10:
> having a T-1 that cause problems

What are the problems?

MASPPrototipi, 2020-07-10:
Print start with error NO TOOL SELECT

Cordiali Saluti,
Spazzini Maicol

MASPPrototipi, 2020-07-10:
And after start code it runs a change tool script. We don’t need a full change script after a start code, but only load the next tool, not unload previous tool and load next tool

Cordiali Saluti,
Spazzini Maicol

lukasmatena, 2020-07-10:
I don't know what printer you use and how your custom gcode looks like, but if you need to take care of the special case of initial_tool==-1, that should be possible by using conditional expressions in the custom gcode. This behaviour is there for years and I don't remember anyone who was unable to do something.

> We don’t need a full change script after a start code

Maybe you don't, but other people with different printers might. Dividing the custom gcode in two might seem simple, but it is a breaking change that would make everyone have to update their profiles. That's why I would rather not change it, unless there is a very compelling reason to do so.

closed 2026-09-29 by prusaslicer-bot[bot]