prusa3d/PrusaSlicer #4507: Won't start, error when start without debug per instructions

xayoz, 2020-07-09:
### Version
_2.3.0_

### Operating system type + version
Windows 10

### Behavior
Followed the "Noob" instructions and after compile get an error code when trying to start "without debug"

[image]

vranisimo, 2020-07-13:
Well, I am facing the same issue
It is probably something with wrong solution configuration, but can't figure it out.

raiscan, 2020-07-14:
Having the same issue, the gui doesn't launch at all

n8bot, 2020-07-14:
It seems like you're trying to build the debug configuration. I've got no experience with that, but building the release configuration works for me.

xayoz, 2020-07-14:
Think I've got it...clicking build configuration in the build menu, then going to without debug seems to work for me.

vranisimo, 2020-07-14:
Ok, so, I followed the "noob" instructions and this part went well.

Without changing any configuration of solution, when I try to Start Debugging or Run without debug, it say:
Unable to start program
...PrusaSlicer\build\src\Release\PrusaSlicer.dll
...PrusaSlicer\build\src\Release\PrusaSlicer.dll is not a valid Win32 application.

I am trying with both Debug and Release options but it's not working.

The important thing is that **I am able** to run program manually, by clicking the prusa-slicer.exe file that is generated as a part of build, but obviously, I am also trying to debug code.

bubnikv, 2020-07-21:
You may need to copy manually the libgmp-10.dll and libmpfr-4.dll next to
where prusaslicer.exe or prusaslicer-console.exe are built.

closed 2026-09-29 by prusaslicer-bot[bot]