prusa3d/PrusaSlicer #4508: Problems with CMake and version_2.3.0-alpha0

Despx72, 2020-07-10:
### Version
version_2.3.0-alpha0

### Operating system type + version
Win7 Professional

Hi all,
I followed Noob guide (step by step) but at the cmake command I stopped with these errors:

************************************************************************
** Visual Studio 2019 Developer Command Prompt v16.6.3
** Copyright (c) 2020 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community>cd c:\PrusaSlicer

c:\PrusaSlicer>cd PrusaSlicer-master

c:\PrusaSlicer\PrusaSlicer-master>mkdir build

c:\PrusaSlicer\PrusaSlicer-master>cd build

c:\PrusaSlicer\PrusaSlicer-master\build>cmake .. -G "Visual Studio 16 2019" -DDE
STDIR="c:\PrusaSlicer\PrusaSlicer-master"
-- Selecting Windows SDK version 10.0.18362.0 to target Windows 6.1.7601.
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: ON
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: ON
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: ON
-- CMAKE_PREFIX_PATH: (default)
Building without Win10 Netfabb STL fixing service support
CMake Error at C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Com
mon7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.17/Modules/FindPac
kageHandleStandardArgs.cmake:164 (message):
  Could NOT find Boost (missing: Boost_INCLUDE_DIR system filesystem thread
  log locale regex chrono atomic date_time)
Call Stack (most recent call first):
  C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/Comm
onExtensions/Microsoft/CMake/CMake/share/cmake-3.17/Modules/FindPackageHandleSta
ndardArgs.cmake:445 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/Comm
onExtensions/Microsoft/CMake/CMake/share/cmake-3.17/Modules/FindBoost.cmake:2145
 (find_package_handle_standard_args)
  CMakeLists.txt:269 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/PrusaSlicer/PrusaSlicer-master/build/CMakeFiles/CMakeOutput.log".
See also "C:/PrusaSlicer/PrusaSlicer-master/build/CMakeFiles/CMakeError.log".

c:\PrusaSlicer\PrusaSlicer-master\build>**

------------------------------------------------------------------------------------------

CMakeError.log:

**Determining if the include file pthread.h exists failed with the following output:
Change Dir: C:/PrusaSlicer/PrusaSlicer-master/build/CMakeFiles/CMakeTmp

Run Build Command(s):C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/MSBuild/Current/Bin/MSBuild.exe cmTC_d0e41.vcxproj /p:Configuration=Debug /p:Platform=x64 /p:VisualStudioVersion=16.0 /v:m && Microsoft (R) Build Engine versione 16.6.0+5ff7b0c9e per .NET Framework

Copyright (C) Microsoft Corporation. Tutti i diritti sono riservati.

  Microsoft (R) C/C++ Optimizing Compiler versione 19.26.28806 per x64

  Copyright (C) Microsoft Corporation. Tutti i diritti  sono riservati.

  cl /c /Zi /W3 /WX- /diagnostics:column /Od /Ob0 /D WIN32 /D _WINDOWS /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /Fo"cmTC_d0e41.dir\Debug\\" /Fd"cmTC_d0e41.dir\Debug\vc142.pdb" /Gd /TC /errorReport:queue "C:\PrusaSlicer\PrusaSlicer-master\build\CMakeFiles\CMakeTmp\CheckIncludeFile.c"

  CheckIncludeFile.c

C:\PrusaSlicer\PrusaSlicer-master\build\CMakeFiles\CMakeTmp\CheckIncludeFile.c(1,10): fatal error C1083: Non è possibile aprire il file inclusione: 'pthread.h': No such file or directory [C:\PrusaSlicer\PrusaSlicer-master\build\CMakeFiles\CMakeTmp\cmTC_d0e41.vcxproj]**

Thanks for your help.
Despx

tamasmeszaros, 2020-07-10:
Check your `CMAKE_PREFIX_PATH` if it points to the right directory of PrusaSlicer dependencies.

Despx72, 2020-07-10:
I apologize, I'm a novice with these procedures (I always started from VS projects and not from generics).
Can you explain better what I have to do?

Many thanks for your patience.

n8bot, 2020-07-11:
It looks like you are using the command to build the dependencies while not in the deps/ folder.

First, cd into /deps and then continue as you were.

bubnikv, 2021-09-22:
https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md

closed 2021-09-22 by bubnikv