prusa3d/PrusaSlicer #4509: OctoPrint Application keys workflow

fredrikbaberg, 2020-07-10:
### Version
2.2.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Creality Ender 3 Pro

### Behavior
When connecting to OctoPrint, PrusaSlicer needs the API key.
From OctoPrint version 1.3.10 there has been an option for Application keys, described as "application specific API key" (https://docs.octoprint.org/en/master/bundledplugins/appkeys.html).
Instead of having the user write/copy the API key this would allow the user to authenticate PrusaSlicer from within OctoPrint. It also allows one to remove the application specific API key in case it has been misplaced.

_Is this a new feature request?_
Yes? Modification to current OctoPrint workflow.

#### Project File (.3MF) where problem occurs
Not relevant

evenprimes, 2020-11-19:
I'm wondering what needs to change. I use application an application specific API key now in Prusa Slicer 2.2 and the 2.3 alpha.

shaver, 2020-11-19:
It would be good if Prusa Slicer used the key-request flow so that the user didn't have to do manual generation and copying.

evenprimes, 2020-11-19:
> It would be good if Prusa Slicer used the key-request flow so that the user didn't have to do manual generation and copying.

Ah, I didn’t know there was a built in flow for that. Yeah, then this request makes sense both for beginners just getting started and folks using multiple systems.

shaver, 2020-11-23:
Yep, see the workflow description here: https://docs.octoprint.org/en/master/bundledplugins/appkeys.html#workflow

closed 2026-09-29 by prusaslicer-bot[bot]