prusa3d/PrusaSlicer #4510: Ironing layer not shown in preview when settings are changed

quojure, 2020-07-10:
2.3.0-alpha0
 7e9e4cb

Windows 10

Weird bug where if I preview a file with ironing turned on - I can see the ironed layer. If I then make any changes to the settings (even unrelated to ironing), the preview removes the ironing layer view. I need to go back into the settings and de-select then select ironing again for it to add back the ironed top layer in preview.

bubnikv, 2020-10-22:
Likely a duplicate of #4510

bubnikv, 2020-10-22:
Hopefully fixed with b1b7a5ad1bd7f07cf94da92c4efbe0dbd0b4a8bc.
Thanks for your report.

closed 2020-10-22 by bubnikv

5teveO, 2020-12-14:
i'd like to report that this bug seems to have reared its head again. If i go and change fill angle from say 45 deg to 135 deg, ironing dissappears, untick ironing, slice and then retick it and slice and its back again.. 

Steve