prusa3d/PrusaSlicer #4511: Cannot change Z position of part/modifier in any way, X/Y position only by dragging

c64club, 2020-07-11:
### Version
2.2.0+, 2.1.0

### Operating system type + version
Windows 10 and others, Linux-x64 (appimage), few different computers

### 3D printer brand / version + firmware version (if known)
all
[image]

### Behavior
Manual repositioning of part/modifier doesn't work. To be precise - always never work in 2.2.0, and never works in 2.1.0. Detailed description: I can manually write a new offset value for eg. X axis but as I press Enter or move cursor to write "position" value in eg. Y axis box, the value in X axis box reverts to previous. So eg after resizing of "box" modifier from default 25mm to 33mm, it hangs "in air" 3mm over table (with eg 7mm, not 3mm, value in "Z axis position") box, and without ability to drop onto table manually or automatically. 
I cannot, let's say, put flat 50x50x10mm "base" and put my 80mm high penguin figure on it. I get a Penguin still sticked to the table, and 50x50x10mm box cutting the figure in half height.

Sometimes  in 2.2.0 (but without significant reason) manual positioning (by writing values) works for particular modifier (eg. 3 parts, 7 modifiers, I can change one box Z position). In this rare case, also "drop 

Almost always, placing a modifier gives strange Z axis position value. eg just after "add modifier" operation, I get 25x25x25 box, sticked to table, but "Z axis position" is 9mm or -0.3mm or any other, looking like random one. No matter "world coordinates" or "local coordinates" are set for its master part.
For "size" and "angle" value boxes, everything works flawlessly. I can rotate and resize with locked or unlocked ratio with no problems.

[image]

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

neophyl, 2020-07-11:
You can pan with either the right or middle mouse buttons on windows machines, not sure how that works on macs.

c64club, 2020-07-12:
Thanks much. Not the "standard" way (point where I need to zoom and scroll) but works fairly convenient on Linux and Windows machines.

closed 2020-07-12 by c64club