prusa3d/PrusaSlicer #4514: What is "Pretty PLA user profile"? Print, filament or printer profile? Where did you get it?

tomservaux, 2020-07-11:
What is "Pretty PLA user profile"? Print, filament or printer profile? Where did you get it?
It always helps, if you can attach as much information to the issue as possible including files.
For this issue, a zipped content of your PrusaSlicer-alpha directory would help.

I am currently not sure what the problem is. We made the system Filament profiles to be suppressed for non-system print or printer profiles. If your print or printer profiles are derived from our profiles, then everything shall work. We have renamed our profiles (see the note on logical names and the "@" sign in the change log), so as of now you will need to edit your user profiles and add the @ character into the names of the "inherits" field. Then everything shall work.

We may want to revise this tightened "compatibility" restriction on system profiles, but it is getting more complicated with the introduction of 3rd party configurations and the logical names of the profiles, and over-complicated rules are likely worse than a bit tighter, but clear rules.

_Originally posted by @bubnikv in https://github.com/prusa3d/PrusaSlicer/issues/3347#issuecomment-568253284_

First of all I want to stress how much I love the changes and improvements you guys have made to the prusaslicer since forking it from Slic3r. It's feature set has convinced me to change over to it from Cura as my main slicer.

I have a problem and judging by the response linked above it sounds as though it is a "works as designed". When I click add/remove filaments under the Filament: dropdown I get this wonderful list of Filament profiles. Then I discover that I can't even use most of them with a Creality printer. 

[image]

We have both Prusa and Creality printers here but when I look at the filament profiles available to us the list is dramatically different between the two. For example, both the Mk3s and the Ender 3 are capable of printing Nylon and yet the Ender 3 has only 3 Creality, 3 Generic and and a couple of Prusament filament profiles (none of which are Nylon) available to it.

This is really frustrating. I see this great, inclusive, thorough list of filament profiles and I cannot even look at them without a text editor unless I have the "correct" printer to use them. Why doesn't the slicer have an option, perhaps in "Expert" mode, to make all filament profiles t available for all printers. I understand that some people will make mistakes if this is the case but In my opinion but please let us make them. As far as I am  concerned, when I click "Expert" I am saying "I got this, thanks" which is exactly what I want. 
I know I can copy/paste and modify. I get it but I am a two man shop and spending time doing this for every material we use is a big bummer.

I am sure you have your reasons as to why you did it this way (another reason I included the link to the other issue response) but I think this would be a great time to get out of the way of people who don't want the slicer deciding which profiles are ok to use for a given piece of equipment. 

Thanks again for all of your hard work on this great software.

Igami, 2020-12-13:
I believe that it would be great if it is possible to inherit profiles from another manufacturer.
This would make it possible to create manufacturer files with only filament and then simply inherit them to each printer or, as @bubnikv mentioned, make the filaments available in the expert settings. There is already a note in the configuration wizard saying "Filaments marked with * are not compatible with some installed printers."

Maybe it would also be easier if there is a printer note for compatible material (e.g. PLA, ABS, Nylon, ...).

LearnIIBurn, 2020-12-27:
Glad I found this as I have the exact same nitpick about Prusa Slicer with my ender 3. I love using Prusa Slicer to try different things and compare results with Cura, but only if I'm printing PLA or PETG. Having no profile, even generic, for TPU, PC, Nylon is somewhat troubling especially when seeing the vast list of profiles that are available. I know it's not your hardware and I hope to own one of your printers someday. But I'm stuck with my Ender 3 for now and even to create my own profile for a specific material type is really quite confusing. Slicer as a whole really is pretty daunting to grasp so that is not really the fault of Prusa. Again I love your slicer and your work on your improved fork, keep up the amazing work.