prusa3d/PrusaSlicer #4516: PrusaSlicer crashes with attached project

0versizedHat, 2020-07-12:
### Version
Version 2.2.0 win64

### Operating system type + version
Microsoft windows 10 pro Version	10.0.19041 Build 19041

### 3D printer brand / version + firmware version (if known)
Mk3s

### Behavior
PrusaSlicer crashes when attempting to slice attached project

#### Project File (.3MF) where problem occurs

[head2.zip](https://github.com/prusa3d/PrusaSlicer/files/4909305/head2.zip)

foreachthing, 2020-07-13:
I don't know what that 3mf contains... but latest build PS crashes:
loglevel9: [head2.txt](https://github.com/prusa3d/PrusaSlicer/files/4910640/head2.txt)

After using anyconv.com (3mf2stl), I got a working stl:
[AnyConv.com__head2.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/4910626/AnyConv.com__head2.stl.zip)

And a 50% smaller file after netfabb fixed it:
[AnyConv.com__head2_fixed.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/4910631/AnyConv.com__head2_fixed.stl.zip)

0versizedHat, 2020-07-13:
Hmm. It's able to open fine for me when using open project, the crash only occurs when trying to slice - I realize I wasn't specific enough in my initial report, forgot to mention that this is an MMU2 project with 5 colors/materials, so it's understandable that converting it to a single STL would reduce the size.
Adding a screenshot of the layout pre-slicing if that helps. Also may be worth noting that it seems like slicing at least partially succeeds, that is, it very briefly displays the sliced preview before crashing within a second-ish, managed to nab a screenshot of that instant, see second pic.
Also, have attached a zip with the individual STLs that make up the composite object, project can be recreated by first importing skin.stl then using 'add part' to add the rest. 
Finally, not sure if it's relevant or not, but it actually slices fine in the initial orientation, it's only when rotating it into the orientation shown in the screenshots that a crash occurs on slicing

[image]
[image]
[stls.zip](https://github.com/prusa3d/PrusaSlicer/files/4910846/stls.zip)

foreachthing, 2020-07-13:
Maybe attach your profile as well. Not everybody has a multi-color printer available ;-)

rtyr, 2021-02-22:
Is this issue still valid? Does it happen with 2.3.0?

I am unable to reproduce it with 2.3.0.

rtyr, 2022-06-20:
No response. Closing.

closed 2022-06-20 by rtyr