prusa3d/PrusaSlicer #4517: Bug? Feature? Why raft are overlapped while print multiple objects?

scott0122, 2020-07-12:
### Version
2.2.0

### Operating system
Windows 10

### Behavior
Raft layers: 3 
First raft layer are overlapped while print multiple objects. And the first overlapped layer cause 2nd and 3rd interface in bad result.

[image]

[image]

[image]

[image]

neophyl, 2020-07-13:
That's what slicer does.  All objects are sliced individually rather than as a plate.  This allows different layer heights on different models etc.
For your particular use case if you load in a single object and then use add instance ( + on numeric keypad) to add in all the duplicates then they will all be part of the same object and so should slice without the rafts overlapping.

When you have multiple different objects that you want to add to the plater where this could be an issue you can use the right click your first object and add function instead of the normal load option, this will load them and treat them as a single object too for slicing purposes.

scott0122, 2020-07-13:
@neophyl 
>> For your particular use case if you load in a single object and then use add instance ( + on numeric keypad) to add in all the duplicates then they will all be part of the same object and so should slice without the rafts overlapping.

I use + to add instances, but first raft layer still overlapped~

[image]
[image]

neophyl, 2020-07-13:
My apologies, when I tried it with some calibration cubes the raft didnt overlap.  I obviously didnt put them close enough (I used auto arrange).  When I manual moved one closer I see that yes it does overlap.
I did try the right click and add part option though and that doesnt overlap the rafts.

scott0122, 2020-07-13:
@neophyl 

I think it should be better not to generate raft "brim". In my experence, it usually cause bad interface result above overlapped area.

This should be better.
[image]

RudolphRiedel, 2020-10-25:
#1275

yoyo42, 2020-12-08:
Still there in alpha3, I guess it isn't going away yet. It seems like a pretty simple thing to fix as it's a layer under all the actual objects, take the convex hull of all the shapes (rubber band bounding box) then draw a raft under that-plus-a-bit. You can still do any per-object layer heights etc because all those happen _above_ the raft, that's sort of the point of it.

RudolphRiedel, 2020-12-08:
Well, there are over 1700 open issues now, I wonder if there is a plan to fix some of these instead of adding more and more features.

scott0122, 2020-12-09:
@yoyo42 @RudolphRiedel 

Right click on arrange icon to adjust gap size. (ver 2.3.0+)

[image]

RudolphRiedel, 2020-12-09:
There are other workarounds as well and so this is only a minor issue.
Yet this remains an issue.

#1275 - I reported this over two years ago

adlerweb, 2021-01-07:
> That's what slicer does. All objects are sliced individually rather than as a plate. This allows different layer heights on different models etc.
Yet when using brim on a print with multiple objects the brim areas are already combined, so there seems to be some kind of collision handling in place for brim (and probably skirt). Just not used for raft atm.

bubnikv, 2021-03-18:
PrusaSlicer 2.4.0 will allow one to disable 1st layer raft expansion.

viper0078, 2021-11-16:
I had high hopes for Ver. 2.4, which would improve the Raft function. However, since the raft still has this problem, I decided that I could not use it.
The figure shows a slice in Simplify3D.
[image]

In Prusa, having to keep the distance between the parts so that the rafts do not collide wastes filament and increases print time.

Prusa has some great features that other slicers don't have, and I'm eager to use it. Unfortunately, I will continue to use S3D until the next version when the raft will be improved.

viper0078, 2021-11-16:
For the production and sale of printed materials, wasted filament and long production times are problems to be avoided.

stepikovo, 2022-08-22:
Duplicate of #2501. As of now, there is no collision detection of individual objects, it is up to the user to prevent them. 
One way how to solve it is to merge the objects.

Closing as a duplicate.

closed 2022-08-22 by stepikovo