prusa3d/PrusaSlicer #4518: Bug: PrusaSlicer 2.2.0 Filament Change Not Working

jackthese, 2020-07-13:
### Version
2.2.0

### Operating system type + version
Windows 10, Version 2004 (latest as of July 13, 2020)

### 3D printer brand / version + firmware version (if known)
Prusa MK 3S firmware 3.9

### Behavior
After I slice, there is no "Orange Ball" option to choose a layer for a filament color change. Left click or right click on the layer bar does nothing. I've verified that 2.1.0 works correctly, so I have uninstalled 2.2.0.

_Is this a new feature request?_

no, this is a bug

rtyr, 2020-07-13:
You mean this? Can you add some screenshot?

[image]

jackthese, 2020-07-15:
Yes, that was exactly the problem, now solved. That Orange Ball with the plus sign did not appear when I right or left-clicked the bar. What solved it was to completely uninstall 2.2.0, reinstall it _**and then I ran the Configuration _**Wizard**. That did the trick, Thanks for following up with me. ...Jack

closed 2020-07-15 by jackthese