prusa3d/PrusaSlicer #4519: infill direction can be unhelpful during first few layers

yoyo42, 2020-07-13:
### Version
2.2.0 +win64

### Operating system type + version
Win 10 x64

### 3D printer brand / version + firmware version (if known)
i3 mk3 converted to mk3s

### Behavior
I have a model with an inlaid design. I print it upside down so the first few layers print the outline of the inlay onto the bed, then change filament for a couple of layers to colour in the inlay, then go back to the original colour.

* _Describe the problem_
The rectilinear infill on the two contrast layers is going same way on both layers and I get gaps between the lines that show all the way through to the backing layers, causing a distinct loss of quality in the contrast layer.

* _Steps needed to reproduce the problem_
Print the attached .3mf with two contracting colours: I had black for the first and last few layers and gold for contrast. I could see black showing through the gold at the line joins

Suggestion: Look for opportunities to merge areas into a continuous layer - both contrast layers could easily be a solid layer of 45 degree lines. Also don't only use +/-45 angles for the complex inlay areas, this will increase the chances of the layers bonding well and filling any gaps.

#### Project File (.3MF) where problem occurs
.3mf and screenshots:
[Tree_Box_Lid.zip](https://github.com/prusa3d/PrusaSlicer/files/4912214/Tree_Box_Lid.zip)
[image]
[image]

rtyr, 2020-07-14:
Both fill and bridging angle can be set in print settings, so you can change it if needed.

yoyo42, 2020-07-15:
That's true, but shouldn't the slicer be ensuring that two layers aren't printed with exactly the same direction anyway?

closed 2026-09-29 by prusaslicer-bot[bot]