prusa3d/PrusaSlicer #4521: Some files are imported empty

LimitlessGreen, 2020-07-13:
### Version
2.2.0

### Operating system type + version
Tested under Windows10 and Manjaro (Arch Linux)

### Behavior
Some files are loaded empty. Example below.

rtyr, 2020-07-14:
It is not empty, the object is just very small. Probably exported with wrong units selected.

This link may help you https://forum.onshape.com/discussion/3870/stl-export-units-issue.

Can we close the issue?

[image]

LimitlessGreen, 2020-07-14:
Well I think this is still an issue, but not an issue, where the files are imported empty.
Your linked forum thread was about Cura, but Cure shows the file correctly.

Changed the title.

rtyr, 2020-07-14:
It doesn't look like your STL was exported in metric scale (or there may be some issue on the OnShape side). I tried to open it in Netfabb, MS 3D builder, Simplify3D, Meshlab, Meshmixer and Cura with same result. Cura just detects it and scale it up, similar to S3D. We may add something similar to PS.

The link was about STL export from OnShape.

Cura
[image]

S3D
[image]

FidelCapo, 2020-07-15:
This feature has already been added and will be part of the next release.

foreachthing, 2020-07-15:
I just tried one of the latest commits and PS does ask about scaling the object.
[image]

Now, I click on Yes:
[image]

But still ... that part is tinyyyyy...!!!
Only 4.4 x 2.79 x 0.3 mm (.17 x .11 x .01 inches) - really?

neophyl, 2020-07-15:
That's because the scaling is just from inches to mm so iirc about 2540%  whatever scale its using on those exports seems to be much much smaller.  From the screen grab above cura was using 100000%.

rtyr, 2020-07-15:
Maybe it is metric after all, exported in meters instead of milimeters scale :-).

foreachthing, 2020-07-15:
Still ... I think the user should know how big his part is/needs to be. And since PS shows how big it is, this should not be an issue.
I think that Cura just takes a guess on how big a part needs to be: if it's very small when *25.4, then it must be *1000 (100000%).

So, I make sure my STL-export is in mm. Even though SolidWorks likes to export in Meters.

rtyr, 2020-07-15:
@foreachthing I am afraid, that guessing is only possible option, because as far as I know STL files contain no scale/unit information.

foreachthing, 2020-07-15:
@rtyr , no they don't!

> (https://en.wikipedia.org/wiki/STL_(file_format) - and the units are arbitrary.)

That's why one has to make sure it's exported correctly... :-\

LimitlessGreen, 2020-07-15:
The guessing of Cura was correct. Here is where the parts are from:
http://forum.banana-pi.org/t/3d-designed-case-for-r2/3906/22

bubnikv, 2021-08-06:
Automatic & manual conversion to / from meters will be implemented in PrusaSlicer 2.4.0-alpha1.

closed 2021-08-06 by bubnikv