prusa3d/PrusaSlicer #4524: RepRapFirmware hotend temp commands not detected in start GCode

emil-e, 2020-07-14:
### Version
2.2.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
BLV mgn cube
Board: Duet WiFi 1.02 or later
Firmware: RepRapFirmware for Duet 2 WiFi/Ethernet 2.04 (2019-11-01b1)
Duet WiFi Server Version: 1.23

### Behavior
PrusaSlicer normally omits temperature commands in the beginning if the appropriate commands are found in the start GCode. However, even if the GCode flavor is set to RepRap/Sprinter, PrusaSlicer will only look for M104/M109 for detecting setting of hotend temperature. These GCodes are deprecated in RepRapFirmware (see for example https://duet3d.dozuki.com/Wiki/Gcode#Section_M104_Set_Extruder_Temperature) and it is instead recommended to use Tx and G10 commands. However, these are not detected.

No particular files are needed to reproduce, the code itself makes it clear that this is not taken into account.

_Is this a new feature request?_
I would consider this a bug.

ardenpm, 2020-10-27:
This should now be fixed by the above referenced pull request which is available in 2.3.0-alpha2. To use the feature you need to change from the `RepRap/Sprinter` firmware flavour to the newly introduced `RepRapFirmware` flavour. The old firmware flavour was relied on for backwards compatibility by some users so a new one was needed (as a side effect the name is now correct as well).

rtyr, 2021-01-29:
Fixed. Closing.

closed 2021-01-29 by rtyr