prusa3d/PrusaSlicer #4526: Adopt Simplify 3D Coast and Wipe capabilities

gnydick, 2020-07-14:
1 Feature request and 1 Enhancement/Change

### PrusaSlicer
**Wipe:** travel over the print while retracting. if 100% of retraction is done before wipe, then no wipe is performed. Cannot wipe like S3D.

### S3D
**Coast:** stop extruding early and continue with movement. extrudes less than standard amount
**Wipe:** travel the print head over existing extrusion traces after extrusion and coasting have finished. can be configured to wipe while retracting like PrusaSlicer, but that isn't the default.

I, and many others, find this subtle distinction to be extremely useful. The fact that PrusaSlicer can't perform these operations means many of us use a different slicer for fine, detailed prints.

rtyr, 2020-07-14:
There is already an issue about coasting https://github.com/prusa3d/PrusaSlicer/issues/246.

gnydick, 2020-07-14:
@rtyr I appreciate that, thanks for linking.

closed 2026-09-29 by prusaslicer-bot[bot]