prusa3d/PrusaSlicer #4529: [Bug] Infill Fill Angle modifier affects top and bottom surfaces

itsthejoker, 2020-07-15:
### Version
Version 2.2.0+linux-x64

### Operating system type + version
Ubuntu 20.04

### 3D printer brand / version + firmware version (if known)
Prusa Mini, stock, 4.1.0-final+1923

### Behavior
When using modifier meshes to apply an **Infill** change using the **Fill Angle** modifier, the top and bottom surfaces are affected.

#### Steps to duplicate and observations

* Enable Expert mode
* Load any STL with flat surfaces on top
* Right click on object in Plater view -> Add Modifier -> choose appropriate shape
* Right click on modifier -> Add Settings -> Infill
* Click Slice Now and observe that the surfaces are at 45 degree angles.
* In the right-hand settings bar, under `Part Settings to modify`, change Fill Angle from 45 to 90 degrees
* Reslice the part
* Observe that the infill has been rotated, but the surfaces have also been effected.

#### Expected Result

The infill has been rotated but the surfaces remain at 45 degree angles.

_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs
N/A

#### Screenshots

Before changing Infill Angle
[image]

After changing Infill Angle
[image]

bubnikv, 2021-06-03:
If I understand correctly, you want the top / bottom infill direction to stay while you only modify sparse internal infill direction?

itsthejoker, 2021-12-27:
I don't know how I missed your reply @bubnikv but I apologize for the very long delay in responding. Yes, that is what I am looking for.

itsthejoker, 2026-09-02:
commenting to keep this open until I can test with PS 3