prusa3d/PrusaSlicer #4530: Missing layers

stuartpb, 2020-07-15:
### Version

2.2.0

### Operating system type + version

Arch Linux, updated a couple days ago

### 3D printer brand / version + firmware version (if known)

Prusa i3 MK3S

### Behavior

[image]

The model on the left is the original; the one on the right was remeshed and repaired in MeshLab. Both have missing layers.

#### Project File (.3MF) where problem occurs

[Felfil_Minotaur.zip](https://github.com/prusa3d/PrusaSlicer/files/4922824/Felfil_Minotaur.zip)

neophyl, 2020-07-15:
I took your remeshed model into Blender and ran the 3d Print tool on it to check for issues.  It reported -
70 Non Manifold Edges and 1971 Intersect Faces.

I fixed it with Blender and sliced it - its the front one.

I've attached my fixed version for you.

[image]

[Felfil_Minotaur_remeshed_neo.zip](https://github.com/prusa3d/PrusaSlicer/files/4923190/Felfil_Minotaur_remeshed_neo.zip)

stuartpb, 2020-07-20:
Do you have a description of the procedure you used to fix the model in Blender?

neophyl, 2020-07-20:
I didn't really have to do much of anything.  I'm currently running the latest (2.82) version of Blender but earlier versions also come with an add on called 3d print tool.  It can be turned on through the preferences option.  Once you enable it, it adds a new tool bar that allows you to run checks on geometry and it can 'attempt' to repair them if you tell it to.  It doesn't always work as its entirely dependent of how broken it is but generally it does a decent job.  Even when it cant the scan it runs can help highlight the areas where there are issues so you can manually fix them.

For your particular model I just clicked check all, which generated the list of errors and then clicked the make manifold button, which just fixed it.  Once fixed I exported the stl and added it to your project , sliced and checked that the model was actually sliced correctly.  

Its probable that the built into Prusa Slicer (on Windows 10) fix through Netfabb option would have worked too or if not on a Windows 10 machine the online version of Netfabb would do the same job.

stuartpb, 2020-07-20:
Actually, I haven't tested it with this particular model, but I'm pretty sure Netfabb is running [MeshFix](https://github.com/MarcoAttene/MeshFix-V2.1) under the hood, and when I tried the MeshFix algorithm it removed entire sections of the model.

neophyl, 2020-07-20:
I was curious so I loaded your original 3mf file into PS2.2.0 and ran the fix through netfabb option on both of the models on the plater.  
It fixed both of them and there don't appear to be any parts missing.

closed 2026-09-29 by prusaslicer-bot[bot]