prusa3d/PrusaSlicer #4532: "Send G-Code to printer host" dialog box blanks out text on Linux

timur-tabi, 2020-07-15:
### Version
2.2.0+linux-x64 appimage

### Operating system type + version
Kubuntu 20.04 Linux

### 3D printer brand / version + firmware version (if known)
MK3 upgraded to MK3s

### Behavior
This is a minor cosmetic bug, but if you don't realize it's a cosmetic bug, it can be disconcerting.

The "Send G-Code to printer host" dialog box includes an option to tell the printer to start printing automatically after upload.  If you click on the option, the filename portion of the g-code file suddenly disappears and you just see ".gcode" in the dialog box.
[image]

[image]

This didn't happen with the previous version on PrusaSlicer.

timur-tabi, 2020-12-28:
FYI, this is still happening with 2.3.0-rc2.

YuSanka, 2021-01-05:
Thanks for your report. I tested PrusaSlicer on Kubuntu 20.04 but I don't observe your issue

[image]

timur-tabi, 2021-01-05:
Which version of the AppImage, the GTK version or the non-GTK version?

YuSanka, 2021-01-06:
Yes, you are right :) I tested it with GTK3. 
And GTk2 really have this issue. And it is a funny issue, because of g-code file name doesn't disappear it just stay white colored, when text control lose the focus and selection (blue background) disappears. And if you just click to the text control the name will be colored as it expected to the black color :).
I'll make some workaround for GTK2 to avoid this misunderstanding.

bubnikv, 2021-01-08:
It should be fixed in PrusaSlicer 2.3.0-rc3.

closed 2021-01-08 by bubnikv

timur-tabi, 2021-01-08:
Confirmed.