prusa3d/PrusaSlicer #4534: Prusa_Slicer_layer_config_ranges.xml object id does not match the 3D/3dmodel.model

Mithrandil, 2020-07-15:
### Version
2.2.0+linux-x64

### Operating system type + version
Kubuntu 20.04

### 3D printer brand / version + firmware version (if known)
does not matter

### Behavior
* When there are multiple instances of multiple objects in the 3mf file, 
and using "Height range Modifier"
the object id in Prusa_Slicer_layer_config_ranges.xml does not match the object id in the 3D/3dmodel.model file

specifically, for example in the attached file, the "Height range Modifier" should refer to `<object id=1>` and `<object id=3>`
(object id=2 is an instance of object id=1, with the same  "Height range Modifier")

3D/3dmodel.model
```
<object id="1" type="model">...</object>
<object id="2" type="model">
  <components>
    <component objectid="1"/>
  </components>
</object>
<object id="3" type="model">...</object>
```
Prusa_Slicer_layer_config_ranges.xml
```
<object id="1">...</object>
<object id="2">...</object> _these ranges are uncorrectly applied to <object id="3" type="model">...</object>_
```

[OPTLH_SupportoVetrinoRovesciato.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4928033/OPTLH_SupportoVetrinoRovesciato.3mf.zip)

[image]

_Is this a new feature request?_
NO, it's a bug

lukasmatena, 2020-07-16:
Actually, we just had a discussion about our 3MF implementation. We will likely rewrite it to fix several issues and improve conformity to the 3MF standard.

Can you confirm that the problem you describe can only be seen when inspecting the file manually? In other words, have you found a scenario where PrusaSlicer loads something different than it previously saved?

Mithrandil, 2020-07-16:
I noticed it just inspecting the file manually 
(or more precisely modifying Prusa_Slicer_layer_config_ranges.xml in the .3mf file using the attached script, which didn't work as expected in case of multiple instances of different objetcs due to this bug in the 3mf implementation)

[NEW-LH-optimizer.zip](https://github.com/prusa3d/PrusaSlicer/files/4931433/NEW-LH-optimizer.zip)

lukasmatena, 2020-07-17:
@Mithrandil Ok, thanks. In that case it's like I have written - we know about it and we will fix it along with more 3MF related problems. I cannot tell when, though. Probably for PrusaSlicer 2.4.0 - it's one of the things that are worth doing right on the first try.

closed 2026-09-29 by prusaslicer-bot[bot]