prusa3d/PrusaSlicer #4538: allow to hide default presets in dropdowns

KubaMiszcz, 2020-07-17:
### Version
version_2.3.0-alpha0

allow to hide default presets (but leave bindings, just hide them in dropdowns). 
I have create my own, based on defaults (pretty well designed), but now sometimes i mistakenly choose default instead my own (I left similar names)

workaround for filaments: I can choose one exotic-name filament that i dont use, then i 'hide' other presets (i cannot choose none filaments)
but for printers it doesnt work, now I'm just choose SLA which i dont have.

rtyr, 2020-07-17:
It is already possible by using compatible printer conditions. For example, if you create new printer profile based on our MK3 and then delete all text in "printer notes", system print profiles will be no longer visible for that printer (you have to delete/modify compatible printer conditions in your custom print profiles of course).

Also, if you need our profiles only as a template, you can temporarily enable option "show incompatible presets" and then click on the flag next to the print profile name.

[image]

rtyr, 2020-07-22:
I believe we can close this.

closed 2020-07-22 by rtyr

KubaMiszcz, 2020-07-23:
yep, your solution works perfectly