prusa3d/PrusaSlicer #4541: Purge Tower Issue

WreckItRalph-OG, 2020-07-17:
### Version
_Version of PrusaSlicer _2.2.0

### Operating system type + version
_What OS are you using, and state any version #s_ Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
PrusaSlicer
Version:   2.2.0+win64
Build:     PrusaSlicer-2.2.0+win64-202003211145

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18363), 64-bit edition
Total RAM size [MB]: 17,105MB
OpenGL installation
GL version:   4.6.0 NVIDIA 441.20
Vendor:       NVIDIA Corporation
Renderer:     GeForce RTX 2070 SUPER/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
[... 346 lines cut ...]
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

### 3D printer brand / version + firmware version Recently purchased a Prusa mk3s with MMU2.... I also have a creality cr10 heavily modified
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_ Marlin Modified by me

### Behavior
* _Describe the problem_ --------- When slicing and using a purgetower, the purge tower is normal sized unless I add supports. then it just grows and is so big that it goes beyond the print area.
* _Steps needed to reproduce the problem_ adding support to any dual color print
  * _If this is a command-line slicing issue, include the options used_ the purge
* _Expected Results_  The purge tower should remain the same size or close to the same size as without support
* _Actual Results_ Purge tower becomes massive and goes beyond print area
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_ Attached screenshots below. Purge volume is the same on both slices, the only parameter changes is adding support

_Is this a new feature request?_ no

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_

[i+love+you+multi+material+picture+frame+design.zip](https://github.com/prusa3d/PrusaSlicer/files/4937638/i%2Blove%2Byou%2Bmulti%2Bmaterial%2Bpicture%2Bframe%2Bdesign.zip)

_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[image][image]
[image]

WreckItRalph-OG, 2020-07-17:
This happens with any multicolor print. regardless  of size.

lukasmatena, 2020-07-17:
This is a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3919, https://github.com/prusa3d/PrusaSlicer/issues/4192 and https://github.com/prusa3d/PrusaSlicer/issues/4361. It is already fixed, the fix will be in 2.3.0 release. There is a workaround until then - see the referenced issues.
Closing.

closed 2020-07-17 by lukasmatena