prusa3d/PrusaSlicer #4546: Implement continual solid (top) layers

ioma8, 2020-07-18:
### Version
2.2

### Operating system type + version
Windows

### 3D printer brand / version + firmware version (if known)
Ender3 

### Behavior 
Now top solid layers are printed in kinda random scattered order. Part of top infill is printed there, part there and this results in visible "joint lines" on tops of surface where single sub-surfaces are joint. Other slicers (eg e3d) print top infill in the most continuous way. Ie start at one corner and fill lines till the end, without skipping to other parts etc. Implement this in PrusaSlicer pls.

_Is this a new feature request?_
yes

rtyr, 2020-07-18:
Monotonous infill will be included in the next PrusaSlicer release. I am not aware of any slicer with such feature, S3D included.

Check this video (from 24:00) https://www.youtube.com/watch?v=NAW2JR-D_QU&t=1443s :-)

ioma8, 2020-07-19:
Hey thx! I didnt know it is in progress! Yes as far as my experience, S3D uses this method as default (without naming it) this is how I discovered better print quality on top infill layer in S3D.
If I understand it correctly, in master git branch its already implemented? Thus if I compile PrusaSlicer from source it will work?

rtyr, 2020-07-19:
Yes, it is already implemented in the master. S3D doesn't use this method, it was most likely only "lucky" case with some particular model.

S3D
[image]

ioma8, 2020-07-20:
Oh, I see, so thats even better!

ioma8, 2020-07-20:
Sadly its impossible to build from source:
[image]

timur-tabi, 2020-09-23:
For the record, "monotonous" is not the right word.  The correct word is "monotonic".

rtyr, 2020-10-17:
Implemented in 2.3.0-alpha1. Closing.

closed 2020-10-17 by rtyr