prusa3d/PrusaSlicer #4549: Feature Request

bheiland, 2020-07-19:
Add ability to store a default values for:
Print Host upload 
Host Type:
HostName IP or URL:
API Key/password.

It would be great if these fields were not filled in and saved, that they could be populated by a default file.
Lets say I have multiple printers that I slice for I could make profiles for each of them and have the printer saved with the profile (how it is currently)

If I have profiles that do not have this information filled out, it could set to a default  .  Example: Octoprint, 111.222.111.222, B432.......

Any unconfigured profile could default to these settings. In my case I only slice over network to one printer so it would be easier to have defaults like this.

bheiland, 2020-09-13:
It seems that Print Host Settings is not in current build?  Is there a branch that might have these features enabled?

rtyr, 2020-09-25:
> 
> 
> It seems that Print Host Settings is not in current build? Is there a branch that might have these features enabled?

[image]

bheiland, 2020-09-30:
Thanks,  I totally missed that!

bubnikv, 2020-10-17:
Implemented in PrusaSlicer 2.3.0-alpha1. Closing.

closed 2020-10-17 by bubnikv