prusa3d/PrusaSlicer #4554: Impossible to build on windows 10

ioma8, 2020-07-20:
### Version
git-master

### Operating system type + version
Windows 10

### Behavior
I follow the "NOOB guide" from docs.
I downloaded git master, entered deps dir, entered this: cmake .. -G "Visual Studio 16 2019"
And I get basically this error:
```
-- Looking for pthread.h - not found
-- Found Threads: TRUE
CMake Error at C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:164 (message):
  Could NOT find Boost (missing: Boost_INCLUDE_DIR system filesystem thread
  log locale regex chrono atomic date_time)
```

I have tried installing boos from source, from boost installer, from vcpkg, nothing helps, it still cant find boost! How do I fix this?
I took day off work to build local PrusaSlicer for me and spent googling and fiddling with this today for about 8 hours already and nothing helped! This needs to get fixed!

krystof1119, 2020-07-21:
I am able to reproduce this issue by running cmake from the `deps` directory, rather than from the `deps/build` directory (therefore the cmake command tries to configure the build of the slicer itself, which fails due to the dependencies not being built by then). Try creating a folder called `build` inside the `deps` folder and running the cmake command from there, this should solve your issue. Also, is the cmake command you specified really what you were running? The guide calls for `cmake .. -G "Visual Studio 16 2019" -DDESTDIR="c:\PrusaSlicer\PrusaSlicer-master"` to be run instead. It also calls for the build folder to be created and used, as I said here.

ioma8, 2020-07-21:
Thank you! Yes I was missing the command and yes I probably wasnt in the build directory (even thought I all the time was convinced I was!)
Now I was able to build the source, bud when it is about to run I get this message box. What is wrong?
[image]

ioma8, 2020-07-21:
When I try to build the prusaslicer_app_gui instead, I get the following return code:
[image]

n8bot, 2020-07-21:
I would recommend simply building from a command line. use the native x64 tools command line, from the Visual Studio directory (run it as admin just because -- hold down shift and right click it and selec to run as admin).

Then do this:
```
cd\C:\Your_PS_Repo_Directory\

mkdir build

cd build

cmake .. -G "Visual Studio 16 2019" -A "x64" -DCMAKE_PREFIX_PATH="C:\[path_to_built_depts]\usr\local"

msbuild /m /P:Configuration=Release ALL_BUILD.vcxproj
```

When that is done, the build should be complete and work! This is assuming you have built the proper deps and you remember what path you built them to.

ioma8, 2020-07-21:
Thank you a lot, it worked! Just before closing this issue I think these commands could be added to the how-to doc.

n8bot, 2020-07-21:
I got them from [SuperSlicer build instructions](https://github.com/supermerill/SuperSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md).

closed 2020-07-21 by ioma8

kmshort, 2020-09-18:
Can you post a quick version of the build instructions that you came up with @ioma8 ?? I'm having trouble and am a noob also.

n8bot, 2020-09-18:
@kmshort I came up with this a while ago and put it together for my own reference. Others have said it worked for them: https://gist.github.com/n8bot/ccf98aa9fef2e3b2cc0647e06ba40cb7

kmshort, 2020-09-18:
> @kmshort I came up with this a while ago and put it together for my own reference. Others have said it worked for them: https://gist.github.com/n8bot/ccf98aa9fef2e3b2cc0647e06ba40cb7

Thanks @n8bot I'm sure it works, but I can't even get cmake to launch, I've installed VS community 2019 with C++ tools.. screw it, too hard. I got the git repo cloned, and the "x64 native tools" env in Administrator mode.. in the /dep dir, and making to a non /PrusaSlicer nested directory, etc. 
bah.. I still need the full idiots guide, I can't even install fecking VS2019 properly lol!

n8bot, 2020-09-18:
IIRC you need to install cmake yourself. VS includes integration tools, but not cmake itself.

kmshort, 2020-09-19:
> IIRC you need to install cmake yourself. VS includes integration tools, but not cmake itself.

Thanks @n8bot cmake is now installed and I can get that far, but I think the title of this issue is appropriate, for me.

But (of course) it fails 

```D:\C_building_stuff\Repos\PrusaSlicer\deps\build>cmake .. -G "Visual Studio 16 2019" -A "x64" -DDESTDIR="D:\C_building_stuff\prusaSlicer_deps"
-- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.19041.
CMake Error at CMakeLists.txt:23 (project):
  Failed to run MSBuild command:

    C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/MSBuild/Current/Bin/MSBuild.exe

  to get the value of VCTargetsPath:

    Microsoft (R) Build Engine version 16.7.0+b89cb5fde for .NET Framework
    Copyright (C) Microsoft Corporation. All rights reserved.

    Build started 19/09/2020 10:17:53.
    Project "D:\C_building_stuff\Repos\PrusaSlicer\deps\build\CMakeFiles\3.18.2\VCTargetsPath.vcxproj" on node 1 (default targets).
    C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.Cpp.Common.props(248,3): error MSB4024: The imported project file "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.Cpp.UnitTest.props" could not be loaded. Root element is missing. [D:\C_building_stuff\Repos\PrusaSlicer\deps\build\CMakeFiles\3.18.2\VCTargetsPath.vcxproj]
    Done Building Project "D:\C_building_stuff\Repos\PrusaSlicer\deps\build\CMakeFiles\3.18.2\VCTargetsPath.vcxproj" (default targets) -- FAILED.

    Build FAILED.

    "D:\C_building_stuff\Repos\PrusaSlicer\deps\build\CMakeFiles\3.18.2\VCTargetsPath.vcxproj" (default target) (1) ->
      C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.Cpp.Common.props(248,3): error MSB4024: The imported project file "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.Cpp.UnitTest.props" could not be loaded. Root element is missing. [D:\C_building_stuff\Repos\PrusaSlicer\deps\build\CMakeFiles\3.18.2\VCTargetsPath.vcxproj]

        0 Warning(s)
        1 Error(s)

    Time Elapsed 00:00:00.11

  Exit code: 1
```

all those files are there.. "Microsoft.Cpp.UnitTest.props".. \deps\build\CMakeFiles\3.18.2\VCTargetsPath.vcxproj]

I won't bother you any more @n8bot , I'm too far out of my depth here. I'm going to assume that someone at my level cannot do it, I need a prusa manual style idiots guide to get it working - just to get the build tools set up.