prusa3d/PrusaSlicer #4555: can't change chinene (linux)

652436962, 2020-07-21:
Version
PrusaSlicer 2.2.0

Operating system type + version
deepin linux 15.11

3D printer brand / version + firmware version (if known)
Prusa i3 MK3s with the last firmware

Behavior
Describe the problem
The software is in French as my operating system

Steps needed to reproduce the problem
Install PrusaSlicer on a computer with the OS in French

Expected Results
Need to change the language of PrusaSlicer in English

Actual Results
No possibility to change the language

lukasmatena, 2020-07-21:
@652436962  What exactly do you mean by "No possibility to change the language"? There is a menu item under Configuration -> Language that is used for that. What is wrong? Is the item not there? Does it do something bad when you click it? What does it do? Does it crash? It doesn't allow you to change the language? Or maybe it does, but the language does not really change? Does it do something else? Your description of the problem is by far not enough.

652436962, 2020-07-21:
> @652436962 What exactly do you mean by "No possibility to change the language"? There is a menu item under Configuration -> Language that is used for that. What is wrong? Is the item not there? Does [...]

I can switch t0 chinese . but It show in English

652436962, 2020-07-29:
I have not do something
now it is ok

rtyr, 2021-02-22:
Is this issue still valid? Does it happen with 2.3.0?

652436962, 2021-03-27:
> 这个问题仍然有效吗？2.3.0是否会发生？

It's no use. I can't switch now

652436962, 2021-03-27:
> Is this issue still valid? Does it happen with 2.3.0?

He still shows English

kang-kang, 2021-12-04:
@652436962  #7410 fixed this bug, it's merged at 2.4.0-beta3.

bubnikv, 2021-12-04:
@kang-kang thanks for heads up. Closing.

closed 2021-12-04 by bubnikv