prusa3d/PrusaSlicer #4556: How Can I use gcode_flavor??

manurafal, 2020-07-21:
### Version
2.2.0

### Operating system type + version
Windows 10 Pro

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro

### Behavior

How can I make the following statement?

{if gcode_flavor=="Marlin"}
xxxxxxxxx
xxxxxxxxx
{endif}

I don't know what I can do if I can use the variable gcode_flavor.

I have tried using it in different sections but I always get error.

### Message Error

Please inspect the file C:\Users\XXXX\AppData\Local\Temp\.3484.gcode.tmp for error messages enclosed between
!!!!! Failed to process the custom G-code template ...
and
!!!!! End of an error report for the custom G-code template ..."

bubnikv, 2020-07-21:
Did you inspect C:\Users\XXXX\AppData\Local\Temp.3484.gcode.tmp ?
I suppose it says "Unknown scalar variable type"

Unfortunately the gcode_flavor is of an enumerated value type, which is not yet supported by the G-code interpreter.

manurafal, 2020-07-24:
thanks to everyone

closed 2020-07-24 by manurafal