prusa3d/PrusaSlicer #4557: Symbolic links in path to user folder cause an error

terribleperson, 2020-07-21:
### Version
Version 2.2.0+win64

### Operating system type + version
Windows 10 Pro Version 1903

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S

### Behavior
When I start PrusaSlicer or attempt to check for updates via 'Configuration -> Check for Updates', I encounter an error dialog. The app doesn't crash, but PrusaSlicer doesn't select any configurations on start up and can't update configurations.
My 'Users' folder located at C:\Users is a symbolic link to D:\WinCSymL\Users. This is done to prevent the many things stored in user folders from taking up space on the small SSD which serves as my C: drive. Symbolic links are generally transparent to user applications; I've only ever seen similar behavior from OpenSCAD. 
This error appears immediately upon starting PrusaSlicer, and PrusaSlicer does not open until the error dialog is closed. An identical error appears if you select 'Ok' in response to the the dialogue about a new configuration update being available on application start up. The application also had a different but similar error on completing the initial setup. 
[image]

bubnikv, 2020-07-22:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/732
It is a boost::filesystem issue. I hoped to solve that issue by switching to std::filesystem, but that is plagued by likely even worse issues.

closed 2020-07-22 by bubnikv

bubnikv, 2020-09-01:
Fixed with 683af51685ec21d09be4c0ef9ac8228b38bcb56d.