prusa3d/PrusaSlicer #4558: Failed to build deps for wxWidgets

Seekerzero, 2020-07-21:
No stash entries found.
CMake Error at D:/PrusaSlicer/PrusaSlicer-master/deps/build/dep_wxWidgets-prefix/tmp/dep_wxWidgets-gitupdate.cmake:189 (message):

  Failed to unstash changes in:
  'D:/PrusaSlicer/PrusaSlicer-master/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets'.

You will have to resolve the conflicts manually

Seekerzero, 2020-07-21:
My PrusaSlicer source code is cloning from the newest master branch.
I tried to build with msbuild first, then get failed with this dependency. And then I tried to build it with nmake, which showed the same result.

n8bot, 2020-07-21:
Hmm this seems related to git. Have you got any changes or anything that you never committed or the like? try doing a git clean.

Seekerzero, 2020-07-21:
let me try. So is that happened even though that the never committed file is from other folder or other project?

n8bot, 2020-07-21:
No clue. Make sure you have the latest version of git, try to clean the repo, and try again. I'm only guessing here. However, the unstash changes thing is definitely git related.

Seekerzero, 2020-07-21:
Thanks! let me update when I solved this problem.

Seekerzero, 2020-07-21:
Here is something wired, the failed unstach changed is inside the /deps/build folder, which is initially empty. it means that the file it mentioned was not from my computer but from the files it downloaded.

Seekerzero, 2020-07-21:
And when I going into the the directory it mentioned:D:/PrusaSlicer/PrusaSlicer-master/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets.
I used git status, it showed me following:

d:\PrusaSlicer\PrusaSlicer-master\deps\build\dep_wxWidgets-prefix\src\dep_wxWidgets>git status
On branch v3.1.3-patched
Your branch is up to date with 'origin/v3.1.3-patched'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)
        modified:   src/expat (modified content)

no changes added to commit (use "git add" and/or "git commit -a")

Seekerzero, 2020-07-21:
And here is the error info from msbuild:

       "d:\PrusaSlicer\PrusaSlicer-master\deps\build\ALL_BUILD.vcxproj" (default target) (1) ->
       "D:\PrusaSlicer\PrusaSlicer-master\deps\build\deps.vcxproj" (default target) (3) ->
       "D:\PrusaSlicer\PrusaSlicer-master\deps\build\dep_wxWidgets.vcxproj" (default target) (21) ->
       (CustomBuild target) ->
         D:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.t
       argets(231,5): error MSB6006: "cmd.exe" exited with code 1. [D:\PrusaSlicer\PrusaSlicer-master\deps\build\dep_w
       xWidgets.vcxproj]

n8bot, 2020-07-21:
My advice: destroy the whole repo, re-clone, and do it methodically. Are there extensive changes you have made, or are you just trying to build master?

Seekerzero, 2020-07-21:
Yes I am trying to build master, and this is definitely clean build.

n8bot, 2020-07-22:
Alright. Start from scratch. Build deps, but send them to a directory outside of the current Repo, so you can re-use them if you want to scrap the repo and start over.

1. Open the x64 Native Tools Command prompt as administrator. 
_(Shift+Right click on the icon->Run as administrator -- running as admin might be unnecessary, but it can't hurt and can be required depending on where you choose to do all of the next steps.)_

2. Build deps.
_(I'm using build2 in case you already have made a build directory. We're making a new one)_
```
cd\C:\[Your_PrusaSlicer_Repo_Path]\

cd deps

mkdir build2

cd build2

cmake .. -G "Visual Studio 16 2019" -A "x64" -DDESTDIR="C:\[empty path where you want to build deps, not inside the repo/directory you are currently in]"
```
cmake will do its thing, then when it completes without error:
```
msbuild /m ALL_BUILD.vcxproj
```
That will build the deps to C:\[empty path where you want to build deps, not inside the repo/directory you are currently in]

I'm pretty sure that DESTDIR path has to not exist. IE, it will create the folder for you, don't make it ahead of time.

3. Then, to build PrusaSlicer, since you'll still be in the deps/build directory, go back to the main repo by doing cd .. twice, like so and continue on:

```
cd..

cd..

mkdir build2

cd build2
```

4. Build PrusaSlicer.
You are now in your C:\[Your_PrusaSlicer_Repo_Path]\build2 directory.
From here, do this, using the exact path you built the deps into, with `\usr\local` added on for the PREFIX_PATH.

```
cmake .. -G "Visual Studio 16 2019" -A "x64" -DCMAKE_PREFIX_PATH="C:\[empty path where you want to build deps, not inside the repo/directory you are currently in]\usr\local"
```
Cmake will do its thing, then to build:

```
msbuild /m /P:Configuration=Release ALL_BUILD.vcxproj
```

And voila, 632 warnings and 0 errors later, you got yourself PrusaSlicer 2.3.0-alpha0.

These instructions are directly from [SuperSlicer build instructions](msbuild /m /P:Configuration=Release ALL_BUILD.vcxproj).

You can now skip building deps whenever you want to build prusaslicer and point to that prefix_path.

Seekerzero, 2020-07-22:
Thanks! let me try first

Seekerzero, 2020-07-22:
Well I followed your instruction, and it failed with the same error for this time. 

       “D:\PrusaSlicer\PrusaSlicer-master\deps\build2\ALL_BUILD.vcxproj”(default target) (1) ->
       “D:\PrusaSlicer\PrusaSlicer-master\deps\build2\deps.vcxproj”(default target) (3) ->
       “D:\PrusaSlicer\PrusaSlicer-master\deps\build2\dep_wxWidgets.vcxproj”(default target) (21) ->
       (CustomBuild target) ->
         D:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.ta
       rgets(231,5): error MSB6006: “cmd.exe”exited，code 1。 [D:\PrusaSlicer\PrusaSlicer-master\deps\build2\dep_wxWidgets.vc
       xproj]

    1927 warning
    1 error

n8bot, 2020-07-22:
Was that on a completely fresh clone of PS? Try that, if not.

Seekerzero, 2020-07-22:
I decompressed the zip file that I downloaded from github this afternoon.

Seekerzero, 2020-07-22:
So should be the clean clone.

n8bot, 2020-07-22:
Was it in the same location as the git repository? Is there a hidden .git folder? Did you actually delete the entire contents of all the folders or just add the files from the zip over top?

Seekerzero, 2020-07-22:
Yes I deleted the entire folder that I extracted. Should delete the hidden .git folder also.

n8bot, 2020-07-22:
In one of your posts, it says you have checked out a branch `v3.1.3-patched` is this correct?

Seekerzero, 2020-07-22:
Yes, that's under PrusaSlicer-master\deps\build\dep_wxWidgets-prefix\src\dep_wxWidgets folder.

Seekerzero, 2020-07-22:
Then 3.1.3-patched should related to wxWidgets dependency.

n8bot, 2020-07-22:
Ok that seems correct. I'm building the deps myself at the moment to troubleshoot this for you. What version of git do you have?

Seekerzero, 2020-07-22:
D:\Program Files (x86)\Microsoft Visual Studio\2019\Community>git --version                                             
git version 2.27.0.windows.1

Seekerzero, 2020-07-22:
Yes, with the version of Visual Studio 2019

n8bot, 2020-07-22:
Well, I'm stumped. Sorry!

Seekerzero, 2020-07-22:
Well, thank you so much! let me try one more time >=<

n8bot, 2020-07-22:
I uploaded the deps that I just built, so that you can just move on and try to build PS itself. [See here](https://github.com/n8bot/PrusaSlicer/releases/tag/version_2.3.0-alpha0).

Seekerzero, 2020-07-22:
Should I just downloaded the zip file or the source code?

Seekerzero, 2020-07-22:
I downloaded the zip file, and plz allow me to take some time to extract it out.

n8bot, 2020-07-22:
Not the source code, just the zip file or 7z file called PrusaSlicer-2.3.0-alpha0_deps_Jul_21_2020

Then, extract that, and point the cmake command for building prusaslicer to wherever you extracted that directory.

Seekerzero, 2020-07-22:
which is the DCMAKE_PREFIX_PATH right?

n8bot, 2020-07-22:
yeah, and add on \usr\local to the end.

It would look like this `D:\temp\prusadeps\usr\local"

Seekerzero, 2020-07-22:
Here is the cmake result, and let me upload this info first, then I will build it:
I:\PrusaSlicer-version_2.3.0-alpha0\build>cmake .. -G "Visual Studio 16 2019" -A "x64" -DCMAKE_PREFIX_PATH="I:\temp\usr\local"
-- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.18363.
-- The C compiler identification is MSVC 19.26.28806.0
-- The CXX compiler identification is MSVC 19.26.28806.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.26.28801/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.26.28801/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: ON
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: ON
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: ON
-- CMAKE_PREFIX_PATH: I:\temp\usr\local (from cache or command line)
CMake Warning at CMakeLists.txt:98 (message):
  CMAKE_PREFIX_PATH element doesn't exist: I:\temp\usr\local

Building without Win10 Netfabb STL fixing service support
-- Looking for pthread.h
-- Looking for pthread.h - not found
-- Found Threads: TRUE
^C
I:\PrusaSlicer-version_2.3.0-alpha0\build>cmake .. -G "Visual Studio 16 2019" -A "x64" -DCMAKE_PREFIX_PATH="I:\temp\prusadeps\usr\local"
-- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.18363.
-- The C compiler identification is MSVC 19.26.28806.0
-- The CXX compiler identification is MSVC 19.26.28806.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.26.28801/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.26.28801/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: ON
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: ON
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: ON
-- CMAKE_PREFIX_PATH: I:\temp\prusadeps\usr\local (from cache or command line)
Building without Win10 Netfabb STL fixing service support
-- Found Boost 1.70.0 at I:/temp/prusadeps/usr/local/lib/cmake/Boost-1.70.0
--   Requested configuration: QUIET REQUIRED COMPONENTS system;filesystem;thread;log;locale;regex;chrono;atomic;date_time
[... 86 lines cut ...]
-- Performing Test LLVM_HAS_ATOMICS
-- Performing Test LLVM_HAS_ATOMICS - Success
-- OpenVDB libraries: I:/temp/prusadeps/usr/local/lib/libopenvdb.lib;I:/temp/prusadeps/usr/local/lib/libopenvdbd.lib
-- Miniz NOT found in system, using bundled version...
-- qhull Version: 7.2.0 (static linking)
-- IGL NOT found, using bundled version...
-- Visual Leak Detector (VLD) is not found.
-- Using header-only CGAL
-- Targetting Visual Studio 16 2019
-- Target build enviroment supports auto-linking
-- Using VC toolset 142.
-- Generator uses intermediate configuration directory: $(Configuration)
-- Found GMP: I:/temp/prusadeps/usr/local/lib/libgmp-10.lib
-- Found MPFR: I:/temp/prusadeps/usr/local/lib/libmpfr-4.lib
-- Found Boost 1.70.0 at I:/temp/prusadeps/usr/local/lib/cmake/Boost-1.70.0
--   Requested configuration: QUIET REQUIRED
-- Found Boost: I:/temp/prusadeps/usr/local/lib/cmake/Boost-1.70.0/BoostConfig.cmake (found suitable version "1.70.0", minimum required is "1.48")
-- Boost include dirs: I:/temp/prusadeps/usr/local/include/boost-1_70
-- Boost libraries:    Boost::system;Boost::filesystem;Boost::thread;Boost::log;Boost::locale;Boost::regex;Boost::chrono;Boost::atomic;Boost::date_time
-- WXWIN environment set to:
-- Found wxWidgets: debug;I:/temp/prusadeps/usr/local/lib/vc_x64_lib/wxmsw31ud_html.lib;optimized;I:/temp/prusadeps/usr/local/lib/vc_x64_lib/wxmsw31u_html.lib;debug;I:/temp/prusadeps/usr/local/lib/vc_x64_lib/wxmsw31ud_adv.lib;optimized;I:/temp/prusadeps/usr/local/lib/vc_x64_lib/wxmsw31u_adv.lib;debug;I:/temp/prusadeps/usr/local/lib/vc_x64_lib/wxmsw31ud_gl.lib;optimized;I:/temp/prusadeps/usr/local/lib/vc_x64_lib/wxmsw31u_gl.lib;debug;I:/temp/prusadeps/usr/local/lib/vc_x64_lib/wxmsw31ud_core.lib;optimized;I:/temp/prusadeps/usr/local/lib/vc_x64_lib/wxmsw31u_core.lib;debug;I:/temp/prusadeps/usr/local/lib/vc_x64_lib/wxbase31ud.lib;optimized;I:/temp/prusadeps/usr/local/lib/vc_x64_lib/wxbase31u.lib;debug;I:/temp/prusadeps/usr/local/lib/vc_x64_lib/wxtiffd.lib;optimized;I:/temp/prusadeps/usr/local/lib/vc_x64_lib/wxtiff.lib;debug;I:/temp/prusadeps/usr/local/lib/vc_x64_lib/wxjpegd.lib;optimized;I:/temp/prusadeps/usr/local/lib/vc_x64_lib/wxjpeg.lib;debug;I:/temp/prusadeps/usr/local/lib/vc_x64_lib/wxregexud.l [... cut ...] al/lib/vc_x64_lib/wxregexud.lib;optimized;I:/temp/prusadeps/usr/local/lib/vc_x64_lib/wxregexu.lib;opengl32;glu32;winmm;comctl32;uuid;oleacc;uxtheme;rpcrt4;shlwapi;version;wsock32;optimized;I:/temp/prusadeps/usr/local/lib/libpng16_static.lib;debug;I:/temp/prusadeps/usr/local/lib/libpng16_staticd.lib;I:/temp/prusadeps/usr/local/lib/zlib.lib;I:/temp/prusadeps/usr/local/lib/expat.lib
-- Configuring done
-- Generating done
-- Build files have been written to: I:/PrusaSlicer-version_2.3.0-alpha0/build

Seekerzero, 2020-07-22:
I changed save root directory from D: to I:

Seekerzero, 2020-07-22:
well also ignore the first part of log, I just found that I put the deps directory wrong so I stop it and clean the build directory for building again.

n8bot, 2020-07-22:
That is the result you want. Move on to the next step.

Seekerzero, 2020-07-22:
It turns out problems with src\slic3r folder.  5981 errors comes out...

Seekerzero, 2020-07-22:
where can I found the msbuild log, so I can send it to you.

n8bot, 2020-07-22:
I do not know. I never really look too deep into the log I just see any errors at the end of the output.

Seekerzero, 2020-07-22:
PrinterPresetCollection”: 非法的成员初始化:“PresetCollection”不是基或成员 (编译源文件 I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3
       r\Utils\PresetUpdater.cpp) [I:\PrusaSlicer-version_2.3.0-alpha0\build\src\slic3r\libslic3r_gui.vcxproj]
         I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\GUI\Preset.hpp(506,20): error C2065: “m_type”: 未声明的标识符 (编译源文件 I
       :\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\Utils\PresetUpdater.cpp) [I:\PrusaSlicer-version_2.3.0-alpha0\buil
       d\src\slic3r\libslic3r_gui.vcxproj]
         I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\GUI\Preset.hpp(507,36): error C2065: “m_presets”: 未声明的标识符 (编译源文
       件 I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\Utils\PresetUpdater.cpp) [I:\PrusaSlicer-version_2.3.0-alpha0\b
       uild\src\slic3r\libslic3r_gui.vcxproj]
         I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\GUI\Preset.hpp(507,56): error C2065: “m_num_default_presets”: 未
       声明的标识符 (编译源文件 I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\Utils\PresetUpdater.cpp) [I:\PrusaSlicer-version_2.
       3.0-alpha0\build\src\slic3r\libslic3r_gui.vcxproj]
         I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\GUI\Preset.hpp(507,79): error C2065: “m_presets”: 未声明的标识符 (编译源文
       件 I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\Utils\PresetUpdater.cpp) [I:\PrusaSlicer-version_2.3.0-alpha0\b
       uild\src\slic3r\libslic3r_gui.vcxproj]
         I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\GUI\Preset.hpp(507,24): error C2672: “std::lower_bound”: 未找 到匹配的
       重载函数 (编译源文件 I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\Utils\PresetUpdater.cpp) [I:\PrusaSlicer-version_2.3.
       0-alpha0\build\src\slic3r\libslic3r_gui.vcxproj]
         I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\GUI\Preset.hpp(507,1): error C2780: “_FwdIt std::lower_bound(_F
       wdIt,_FwdIt,const _Ty &)”: 应输入 3 个参数，却提供了 2 个 (编译源文件 I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\Utils\Preset
       Updater.cpp) [I:\PrusaSlicer-version_2.3.0-alpha0\build\src\slic3r\libslic3r_gui.vcxproj]
         I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\GUI\Preset.hpp(507,1): error C2780: “_FwdIt std::lower_bound(_F
       wdIt,const _FwdIt,const _Ty &,_Pr)”: 应输入 4 个参数，却提供了 2 个 (编译源文件 I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\Ut
       ils\PresetUpdater.cpp) [I:\PrusaSlicer-version_2.3.0-alpha0\build\src\slic3r\libslic3r_gui.vcxproj]
         I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\GUI\Preset.hpp(508,1): error C3536: “it”: 初始化之前无法使用 (编译源文件 I:\P
       rusaSlicer-version_2.3.0-alpha0\src\slic3r\Utils\PresetUpdater.cpp) [I:\PrusaSlicer-version_2.3.0-alpha0\build\s
       rc\slic3r\libslic3r_gui.vcxproj]
         I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\GUI\Preset.hpp(508,19): error C2065: “m_presets”: 未声明的标识符 (编译源文
       件 I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\Utils\PresetUpdater.cpp) [I:\PrusaSlicer-version_2.3.0-alpha0\b
       uild\src\slic3r\libslic3r_gui.vcxproj]
         I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\GUI\Preset.hpp(510,36): error C2065: “m_num_default_presets”: 未
       声明的标识符 (编译源文件 I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\Utils\PresetUpdater.cpp) [I:\PrusaSlicer-version_2.
       3.0-alpha0\build\src\slic3r\libslic3r_gui.vcxproj]
         I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\GUI\Preset.hpp(511,21): error C2065: “m_presets”: 未声明的标识符 (编译源文
       件 I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\Utils\PresetUpdater.cpp) [I:\PrusaSlicer-version_2.3.0-alpha0\b
       uild\src\slic3r\libslic3r_gui.vcxproj]
         I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\GUI\Preset.hpp(512,26): error C2065: “m_presets”: 未声明的标识符 (编译源文
       件 I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\Utils\PresetUpdater.cpp) [I:\PrusaSlicer-version_2.3.0-alpha0\b
       uild\src\slic3r\libslic3r_gui.vcxproj]
         I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\GUI\Preset.hpp(516,1): error C2440: “return”: 无法从“int”转 换为“std::
       _Deque_iterator<std::_Deque_val<std::_Deque_simple_types<_Ty>>>” [I:\PrusaSlicer-version_2.3.0-alpha0\build\src\
       slic3r\libslic3r_gui.vcxproj]
         I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\GUI\Preset.hpp(519,1): error C2355: “this”: 只能在非静态成员函数或非静态数据成员初
       始值设定项的内部引用 (编译源文件 I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\Utils\PresetUpdater.cpp) [I:\PrusaSlicer-versio
       n_2.3.0-alpha0\build\src\slic3r\libslic3r_gui.vcxproj]
         I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\GUI\Preset.hpp(521,24): error C2065: “m_map_system_profile_rena
       med”: 未声明的标识符 (编译源文件 I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\Utils\PresetUpdater.cpp) [I:\PrusaSlicer-ver
       sion_2.3.0-alpha0\build\src\slic3r\libslic3r_gui.vcxproj]
         I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\GUI\Preset.hpp(522,1): error C3536: “it_renamed”: 初始化之前无法使用 (编译
       源文件 I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\Utils\PresetUpdater.cpp) [I:\PrusaSlicer-version_2.3.0-alpha0
       \build\src\slic3r\libslic3r_gui.vcxproj]
         I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\GUI\Preset.hpp(522,31): error C2065: “m_map_system_profile_rena
       med”: 未声明的标识符 (编译源文件 I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\Utils\PresetUpdater.cpp) [I:\PrusaSlicer-ver
       sion_2.3.0-alpha0\build\src\slic3r\libslic3r_gui.vcxproj]
         I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\GUI\Preset.hpp(522,69): error C2065: “m_presets”: 未声明的标识符 (编译源文
[... cut ...]
         I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r/GUI/PresetBundle.hpp(71,81): error C2065: “printers”: 未声明的标识符 (
       编译源文件 I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\Utils\PresetUpdater.cpp) [I:\PrusaSlicer-version_2.3.0-alph
       a0\build\src\slic3r\libslic3r_gui.vcxproj]
         I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r/GUI/PresetBundle.hpp(71,9): fatal error C1003: 错误计数超过 100 ；正在停止编
       译 (编译源文件 I:\PrusaSlicer-version_2.3.0-alpha0\src\slic3r\Utils\PresetUpdater.cpp) [I:\PrusaSlicer-version_2.3.0-a
       lpha0\build\src\slic3r\libslic3r_gui.vcxproj]

    566 个警告
    5981 个错误

Seekerzero, 2020-07-22:
I am sure it was said that there are unrecognized grammar and illegal initialization for src\slic3r folder. sorry for the confusion of Chinese.

n8bot, 2020-07-22:
I'm very puzzled. The only thing I could think of is some tool or something missing in Visual Studio. When I installed it, I followed the instruction to "select every option for C++ desktop development workload" and so I selected every option for that workload. See if that helps you. Google about selecting workload options after installing Visual Studio.

Seekerzero, 2020-07-22:
OK, let me try.

n8bot, 2020-07-22:
I think those errors indicate something has been changed in the source. I don't know if you're using the right repository. Are you sure it's the master branch of the Prusa3D PrusaSlicer repo?

Seekerzero, 2020-07-22:
I first used the master branch through clone, it shows these errors, then I tried 2.30 alpha release on May 5th, which showed the same result.

n8bot, 2020-07-22:
Just ensure you are using the most up-to date master branch, and clear everything out and follow the instructions exactly (using the deps I provided) and there is no way it shouldn't work! Check those SuperSlicer build instructions I linked to earlier, they are the source of my knowledge on this topic.

Seekerzero, 2020-07-22:
OK, I would try again!

Seekerzero, 2020-07-22:
I tried twice with msbuild and one time with nmake, they all shows the same error when "Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/AboutDialog.cpp.obj"
It seems like there are some libraries is missing, it can't recognize a lot of definition. And I am sure I used the latest version from master.

Seekerzero, 2020-07-22:
[ 60%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/AboutDialog.cpp.obj                              AboutDialog.cpp                                                                                                         I:\PrusaSlicer\src\slic3r\GUI\AboutDialog.cpp(1): warning C4819: 该文件包含不能在当前代码页(936)中表示的字符。请将该文件保存为 Unicode 格式以防止数据丢失                                                                                       I:\PrusaSlicer\src\slic3r\GUI\GUI.hpp(1): warning C4819: 该文件包含不能在当前代码页(936)中表示的字符。请将该文件保存为 Unicode 格式以防止数据丢失                                                                                               I:\PrusaSlicer\src\slic3r\GUI\GUI_App.hpp(1): warning C4819: 该文件包含不能在当前代码页(936)中表示的字符。请将该文件保存为 Unicode 格式以防止数据丢失                                                                                           I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(1): warning C4819: 该文件包含不能在当前代码页(936)中表示的字符。请将该文件保存 为 Unicode 格式以防止数据丢失                                                                                           I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(477): error C2001: 常量中有换行符                                              I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(574): error C2504: “Slic3r::PresetCollection”: 未定义基类                    I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(578): error C3668: “Slic3r::PresetCollection::PrinterPresetCollection::default_preset_for”: 包含重写说明符“override”的方法没有重写任何基类方法                                                     I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(583): error C2059: 语法错误:“namespace”                                      I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(583): error C2334: “{”的前面有意外标记；跳过明显的函数体                     I:\PrusaSlicer\src\imgui\imconfig.h(101): error C2059: 语法错误:“namespace”                                           I:\PrusaSlicer\src\imgui\imconfig.h(102): error C2143: 语法错误: 缺少“;”(在“{”的前面)                               I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(274): error C2065: “m_presets”: 未声明的标识符                               I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(274): error C2065: “m_num_default_presets”: 未声明的标识符                   I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(275): error C2065: “m_presets”: 未声明的标识符                               I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(275): error C2065: “m_num_default_presets”: 未声明的标识符                   I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(276): error C2065: “m_presets”: 未声明的标识符                               I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(276): error C2065: “m_num_default_presets”: 未声明的标识符                   I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(277): error C2065: “m_presets”: 未声明的标识符                               I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(278): error C2065: “m_presets”: 未声明的标识符                               I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(279): error C2065: “m_presets”: 未声明的标识符                               I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(283): error C2065: “m_type”: 未声明的标识符                                  I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(288): error C2065: “m_presets”: 未声明的标识符                               I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(332): error C2065: “m_bitmap_compatible”: 未声明的标识符                     I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(333): error C2065: “m_bitmap_incompatible”: 未声明的标识符                   I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(334): error C2065: “m_bitmap_lock”: 未声明的标识符                           I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(335): error C2065: “m_bitmap_lock_open”: 未声明的标识符                      I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(339): error C2065: “m_default_suppressed”: 未声明的标识符                    I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(344): error C2065: “m_presets”: 未声明的标识符                               I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(344): error C2065: “m_idx_selected”: 未声明的标识符                          I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(345): error C2065: “m_presets”: 未声明的标识符                               I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(345): error C2065: “m_idx_selected”: 未声明的标识符                          I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(346): error C2065: “m_idx_selected”: 未声明的标识符                          I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(348): error C2065: “m_idx_selected”: 未声明的标识符                          I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(358): error C2065: “m_edited_preset”: 未声明的标识符                         I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(359): error C2065: “m_edited_preset”: 未声明的标识符                         I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(369): error C2065: “m_presets”: 未声明的标识符                               I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(370): error C2065: “m_idx_selected”: 未声明的标识符                          I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(374): error C2065: “m_num_default_presets”: 未声明的标识符                   I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(374): error C2065: “m_presets”: 未声明的标识符                               I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(375): error C2065: “m_num_default_presets”: 未声明的标识符                   I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(375): error C2065: “m_presets”: 未声明的标识符                               I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(378): error C2065: “m_idx_selected”: 未声明的标识符                          I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(378): error C2065: “m_edited_preset”: 未声明的标识符                         I:\PrusaSlicer\src\ [... cut ...]  fatal error U1077: “"D:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\HostX64\x64\nmake.exe"”: 返回代码“0x2”                                                                             Stop.                                                                                                                   NMAKE : fatal error U1077: “"D:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\HostX64\x64\nmake.exe"”: 返回代码“0x2”

Seekerzero, 2020-07-22:
Here is the version with better format
[  1%] Built target admesh
[  2%] Built target nowide
[  3%] Built target encoding-check
[  3%] Checking source files encodings for target avrdude
[  3%] Built target encoding-check-avrdude
[  3%] Checking source files encodings for target avrdude-slic3r
[  3%] Built target encoding-check-avrdude-slic3r
[  3%] Checking source files encodings for target semver
[  3%] Built target encoding-check-semver
[  3%] Checking source files encodings for target libslic3r
[  3%] Built target encoding-check-libslic3r
[  4%] Checking source files encodings for target libslic3r_gui
[  4%] Built target encoding-check-libslic3r_gui
[  4%] Built target global-encoding-check
[  4%] Built target avrdude-conf-gen
[  4%] Built target gen_conf_h
[ 14%] Built target avrdude
[ 15%] Built target clipper
[ 24%] Built target qhullstatic
[ 24%] Built target polypartition
[ 26%] Built target poly2tri
[ 26%] Built target miniz_static
[ 29%] Built target glu-libtess
[ 30%] Built target semver
[ 31%] Built target imgui
[ 32%] Built target hidapi
[ 33%] Built target libnest2d
[ 33%] Built target libslic3r_cgal
[ 60%] Built target libslic3r
[ 60%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/AboutDialog.cpp.obj
AboutDialog.cpp
I:\PrusaSlicer\src\slic3r\GUI\AboutDialog.cpp(1): warning C4819: 该文件包含不能在当前代码页(936)中表示的字符。请将该文件保存为 Unicode 格式以防止数据丢失
I:\PrusaSlicer\src\slic3r\GUI\GUI.hpp(1): warning C4819: 该文件包含不能在当前代码页(936)中表示的字符。请将该文件保存为 Unicode 格式以防止数据丢失
I:\PrusaSlicer\src\slic3r\GUI\GUI_App.hpp(1): warning C4819: 该文件包含不能在当前代码页(936)中表示的字符。请将该文件保存为 Unicode 格式以防止数据丢失
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(1): warning C4819: 该文件包含不能在当前代码页(936)中表示的字符。请将该文件保存 为 Unicode 格式以防止数据丢失
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(477): error C2001: 常量中有换行符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(574): error C2504: “Slic3r::PresetCollection”: 未定义基类
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(578): error C3668: “Slic3r::PresetCollection::PrinterPresetCollection::default_preset_for”: 包含重写说明符“override”的方法没有重写任何基类方法
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(583): error C2059: 语法错误:“namespace”
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(583): error C2334: “{”的前面有意外标记；跳过明显的函数体
I:\PrusaSlicer\src\imgui\imconfig.h(101): error C2059: 语法错误:“namespace”
I:\PrusaSlicer\src\imgui\imconfig.h(102): error C2143: 语法错误: 缺少“;”(在“{”的前面)
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(274): error C2065: “m_presets”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(274): error C2065: “m_num_default_presets”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(275): error C2065: “m_presets”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(275): error C2065: “m_num_default_presets”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(276): error C2065: “m_presets”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(276): error C2065: “m_num_default_presets”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(277): error C2065: “m_presets”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(278): error C2065: “m_presets”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(279): error C2065: “m_presets”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(283): error C2065: “m_type”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(288): error C2065: “m_presets”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(332): error C2065: “m_bitmap_compatible”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(333): error C2065: “m_bitmap_incompatible”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(334): error C2065: “m_bitmap_lock”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(335): error C2065: “m_bitmap_lock_open”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(339): error C2065: “m_default_suppressed”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(344): error C2065: “m_presets”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(344): error C2065: “m_idx_selected”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(345): error C2065: “m_presets”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(345): error C2065: “m_idx_selected”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(346): error C2065: “m_idx_selected”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(348): error C2065: “m_idx_selected”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(358): error C2065: “m_edited_preset”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(359): error C2065: “m_edited_preset”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(369): error C2065: “m_presets”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(370): error C2065: “m_idx_selected”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(374): error C2065: “m_num_default_presets”: 未声明的标识符
[... 78 lines cut ...]
        [
            _Ty=Slic3r::Preset
        ]
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(516): note: 无构造函数可以接受源类型，或构造函数重载决策不明确
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(519): error C2355: “this”: 只能在非静态成员函数或非静态数据成员初始值设定项的内部引用
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(521): error C2065: “m_map_system_profile_renamed”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(522): error C3536: “it_renamed”: 初始化之前无法使用
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(522): error C2065: “m_map_system_profile_renamed”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(522): error C2065: “m_presets”: 未声明的标识符
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(522): error C2355: “this”: 只能在非静态成员函数或非静态数据成员初始值设定项的内部引用
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(523): error C2065: “m_map_system_profile_renamed”: 未声明的标识符
[... cut ...]
I:\PrusaSlicer\src\slic3r\GUI\Preset.hpp(524): fatal error C1003: 错误计数超过 100；正在停止编译
NMAKE : fatal error U1077: “"D:\PROGRA~1\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\Hostx64\x64\cl.exe"”: 返回代码“0x2”
Stop.
NMAKE : fatal error U1077: “"D:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\HostX64\x64\nmake.exe"”: 返回代码“0x2”
Stop.
NMAKE : fatal error U1077: “"D:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\HostX64\x64\nmake.exe"”: 返回代码“0x2”
Stop.

Seekerzero, 2020-07-22:
Well let's look at this tomorrow. rest is also important.

lukasmatena, 2020-07-22:
@Seekerzero I have not read the whole thread (sorry), but the git message that you posted in https://github.com/prusa3d/PrusaSlicer/issues/4558#issuecomment-662149893 indicates that there are local modification to expat. They may be in the way when the build script tries to update, I suggest getting rid of them. Note that your git might be configured to do some kind of changes automatically (such as CR/CRLF line endings or something), so maybe it did the changes behind your back.

Seekerzero, 2020-07-22:
@lukasmatena  So could you please give me any instruction to do that. I am not very familiar with git. Thanks!

lukasmatena, 2020-07-22:
I would start by moving into that directory and trying `git diff` to see what the difference is.

Seekerzero, 2020-07-22:
I mean disable the automatically change from git. I am pretty sure that the edit was made during the process of building dependency. So it probably won't allow me to use git diff before the start of building.

lukasmatena, 2020-07-22:
You could have used `git status`, you can use `git diff` as well (I think).
But note that I do not have time to focus on this fully. This does not seem like a problem with PrusaSlicer.

n8bot, 2020-07-22:
Is it possible that you are accidentally (still) using a zip file of the source with the tag `version-2.3.0-alpha0`? I think that code is much older than what is in the current master. This is obviously a git issue/issue with your configuration.

Do you perhaps not have the right version of Cmake? I know less about cmake than I do about git so I'm stabbing in the dark here. However, maybe this issue should be closed so that it doesn't occupy (as much) time of the devs. It seems unrelated to PrusaSlicer directly.

Seekerzero, 2020-07-23:
My Cmake version is 3.18 and I am pretty sure that I clone the latest master for compiling. 
Anyway, let's close this. If I have any problem, let me ask later.

closed 2020-07-23 by Seekerzero

Seekerzero, 2020-07-24:
@n8bot Hi sir, I might found the reason why I failed to build PrusaSlicer, the PkgConfig is not found in your environment. Do you know what is this? The related log is bold below.
 
I:\PrusaSlicer\build>cmake .. -G "Visual Studio 16 2019" -A "x64" -DCMAKE_PREFIX_PATH="I:\temp\prusadeps\usr\local"
-- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.18363.
-- The C compiler identification is MSVC 19.26.28806.0
-- The CXX compiler identification is MSVC 19.26.28806.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.26.28801/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.26.28801/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: ON
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: ON
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: ON
-- CMAKE_PREFIX_PATH: I:\temp\prusadeps\usr\local (from cache or command line)
Building without Win10 Netfabb STL fixing service support
-- Found Boost 1.70.0 at I:/temp/prusadeps/usr/local/lib/cmake/Boost-1.70.0
--   Requested configuration: QUIET REQUIRED COMPONENTS system;filesystem;thread;log;locale;regex;chrono;atomic;date_time
-- Found boost_headers 1.70.0 at I:/temp/prusadeps/usr/local/lib/cmake/boost_headers-1.70.0
-- Found boost_system 1.70.0 at I:/temp/prusadeps/usr/local/lib/cmake/boost_system-1.70.0
--   libboost_system-vc142-mt-gd-x64-1_70.lib
--   libboost_system-vc142-mt-x64-1_70.lib
-- Adding boost_system dependencies: headers
-- Found boost_filesystem 1.70.0 at I:/temp/prusadeps/usr/local/lib/cmake/boost_filesystem-1.70.0
--   libboost_filesystem-vc142-mt-gd-x64-1_70.lib
--   libboost_filesystem-vc142-mt-x64-1_70.lib
-- Adding boost_filesystem dependencies: headers
-- Found boost_thread 1.70.0 at I:/temp/prusadeps/usr/local/lib/cmake/boost_thread-1.70.0
--   libboost_thread-vc142-mt-gd-x64-1_70.lib
--   libboost_thread-vc142-mt-x64-1_70.lib
-- Adding boost_thread dependencies: chrono;headers
-- Found boost_chrono 1.70.0 at I:/temp/prusadeps/usr/local/lib/cmake/boost_chrono-1.70.0
--   libboost_chrono-vc142-mt-gd-x64-1_70.lib
--   libboost_chrono-vc142-mt-x64-1_70.lib
-- Adding boost_chrono dependencies: headers
-- Found boost_log 1.70.0 at I:/temp/prusadeps/usr/local/lib/cmake/boost_log-1.70.0
--   libboost_log-vc142-mt-gd-x64-1_70.lib
--   libboost_log-vc142-mt-x64-1_70.lib
-- Adding boost_log dependencies: atomic;chrono;date_time;filesystem;regex;thread;headers
-- Found boost_atomic 1.70.0 at I:/temp/prusadeps/usr/local/lib/cmake/boost_atomic-1.70.0
--   libboost_atomic-vc142-mt-gd-x64-1_70.lib
--   libboost_atomic-vc142-mt-x64-1_70.lib
-- Adding boost_atomic dependencies: headers
-- Found boost_date_time 1.70.0 at I:/temp/prusadeps/usr/local/lib/cmake/boost_date_time-1.70.0
--   libboost_date_time-vc142-mt-gd-x64-1_70.lib
--   libboost_date_time-vc142-mt-x64-1_70.lib
-- Adding boost_date_time dependencies: headers
-- Found boost_regex 1.70.0 at I:/temp/prusadeps/usr/local/lib/cmake/boost_regex-1.70.0
--   libboost_regex-vc142-mt-gd-x64-1_70.lib
--   libboost_regex-vc142-mt-x64-1_70.lib
-- Adding boost_regex dependencies: headers
-- Found boost_locale 1.70.0 at I:/temp/prusadeps/usr/local/lib/cmake/boost_locale-1.70.0
--   libboost_locale-vc142-mt-gd-x64-1_70.lib
--   libboost_locale-vc142-mt-x64-1_70.lib
[... 52 lines cut ...]
-- OpenVDB libraries: I:/temp/prusadeps/usr/local/lib/libopenvdb.lib;I:/temp/prusadeps/usr/local/lib/libopenvdbd.lib
-- Miniz NOT found in system, using bundled version...
-- qhull Version: 7.2.0 (static linking)
-- IGL NOT found, using bundled version...
-- Visual Leak Detector (VLD) is not found.
-- Using header-only CGAL
-- Targetting Visual Studio 16 2019
-- Target build enviroment supports auto-linking
-- Using VC toolset 142.
-- Generator uses intermediate configuration directory: $(Configuration)
-- Found GMP: I:/temp/prusadeps/usr/local/lib/libgmp-10.lib
-- Found MPFR: I:/temp/prusadeps/usr/local/lib/libmpfr-4.lib
-- Found Boost 1.70.0 at I:/temp/prusadeps/usr/local/lib/cmake/Boost-1.70.0
--   Requested configuration: QUIET REQUIRED
-- Found Boost: I:/temp/prusadeps/usr/local/lib/cmake/Boost-1.70.0/BoostConfig.cmake (found suitable version "1.70.0", minimum required is "1.48")
-- Boost include dirs: I:/temp/prusadeps/usr/local/include/boost-1_70
-- Boost libraries:    Boost::system;Boost::filesystem;Boost::thread;Boost::log;Boost::locale;Boost::regex;Boost::chrono;Boost::atomic;Boost::date_time
-- Adding precompiled header pchheader.hpp to target libslic3r.
-- WXWIN environment set to:
-- Found wxWidgets: debug;I:/temp/prusadeps/usr/local/lib/vc_x64_lib/wxmsw31ud_html.lib;optimized;I:/temp/prusadeps/usr/local/lib/vc_x64_lib/wxmsw31u_html.lib;debug;I:/temp/prusadeps/usr/local/lib/vc_x64_lib/wxmsw31ud_adv.lib;optimized;I:/temp/prusadeps/usr/local/lib/vc_x64_lib/wxmsw31u_adv.lib;debug;I:/temp/prusadeps/usr/local/lib/vc_x64_lib/wxmsw31ud_gl.lib;optimized;I:/temp/pru [... cut ...] mp/prusadeps/usr/local/lib/vc_x64_lib/wxregexu.lib;opengl32;glu32;winmm;comctl32;uuid;oleacc;uxtheme;rpcrt4;shlwapi;version;wsock32;optimized;I:/temp/prusadeps/usr/local/lib/libpng16_static.lib;debug;I:/temp/prusadeps/usr/local/lib/libpng16_staticd.lib;I:/temp/prusadeps/usr/local/lib/zlib.lib;I:/temp/prusadeps/usr/local/lib/expat.lib
-- Adding precompiled header pchheader.hpp to target libslic3r_gui.
-- Configuring done
-- Generating done
-- Build files have been written to: I:/PrusaSlicer/build

n8bot, 2020-07-24:
I do not know what that package is. I’m just a user of prusaslicer, and I’m new to compiling software too. 

But regardless, the cmake generation from the log you posted was successful. There is no problem.

[11 more replies]