prusa3d/PrusaSlicer #4559: Ironing gives an error when used with height range modifiers

djr7C4, 2020-07-21:
### Version
2.3.0-alpha0

### Operating system type + version
Arch Linux, kernel 5.7.8 x86_64

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
Attempting to enable ironing selectively using a height range modifier causes an error (see the attached screenshot and trace) 

To reproduce:
1. Open box.3mf
2. Right click on the box and click on "Height range Modifier"
3. Click on the gear next to the "Range 0.00 - 2.00 (mm)" object that appears and select "Add settings -> Ironing"
4. Check the box for "Enable ironing" on the dialog the appears
5. At this point, an assertion fails and a backtrace is available (see attached files).
[files.zip](https://github.com/prusa3d/PrusaSlicer/files/4957023/files.zip)
[image]
[backtrace.log](https://github.com/prusa3d/PrusaSlicer/files/4957024/backtrace.log)

lukasmatena, 2020-07-22:
This should be fixed by https://github.com/prusa3d/PrusaSlicer/commit/435355adfe425243e2a5780f1869cff17eeaf05a (thanks to @rongith). Could you please retest? Thanks.

djr7C4, 2020-07-23:
I have tested this against the current master branch (2f43c1f) and it works on the trivial example I provided as well as more sophisticated cases and it seems to work correctly now.

closed 2020-07-23 by djr7C4