prusa3d/PrusaSlicer #4561: Add holes XY compensation field

ioma8, 2020-07-22:
### Version
2.3.0-alpha0

### Operating system type + version
Windows 10 x64 

### Behavior
* SuperSlicer has this amazing feature with key **hole_sizes_compensation**.
* This feature is like XY compensation, but only applies to holes. 
* I find this feature extremely useful because usually I do not want to shring all perimeters, just the holes where screws etc should fit.
* I am not great programmer but if you search hole_sizes_compensation in src/libslic3r in SuperSlicer repository you will find this is only quite few lines of code so should not be hard to merge.
* I am attaching screenshot of where it would be found in [SuperSlicer]: https://github.com/supermerill/SuperSlicer/ 
* Could you please implement this?

_Is this a new feature request?_
yes

[image]

lonelymyp, 2021-02-25:
Is the size of the holes involved in the compensation calculation? 
It seems to me that small holes require more compensation than large holes. A 20mm hole usually has much smaller problems with a 2mm hole.

I suppose there should be some kind of calibration piece with a dozen holes in 1 mm increments. 
After printing, you need to measure the actual dimensions and calculate the compensation coefficient.
And also it is probably necessary to indicate the maximum hole size after which no compensation is required.

spiky2021, 2021-03-04:
Hallo together,
the answer to :
> Is the size of the holes involved in the compensation calculation?
It seems to me that small holes require more compensation than large holes. A 20mm hole usually has much smaller problems with a 2mm hole.

is no, I believe!

There was an old Slic3r compensation algorithm called ArcCompenstion:
https://reprap.org/wiki/ArcCompensation
This was only 2D and relates to a fixed r (or track central Rc). Its results are nearly constant compensations and it showes a dependece on the hole radius R only for values below 1mm. Since people complained to large holes radii with it, it was removed.

In reality such modelling has to be splitted in a constant and a R dependent effect.
A constant compensation is needed for none wrong extrusion factors, z-wobbling and filament diameter variations.
- Technically in many cases the extrusion factor can't be changed, e.g. you need to print liquid tight objects. 
- Similar is true for z-wobbling, maybe buying a new printer helps a little.
- For the diameter it is close to avoidable, just buy another filament.

A real R dependent compensation needs to done as a 3D calculation including the extruded volumes.
The picture below shows a cut through an extruded track.
[image]
Now one has to imaging, that the extruded track is circular bend or in other words the above surface is taken to a rotation integral.
No worry integrations are not needed, just such a track has to visualized. 
Splitting this track by its central radius Rc leads to two volumes each build of a ring and the half of a torus.
The inner ring volume is related to a radius Ri and Rc as:
Vri = pi * (Rc^2 - Ri^2) .
The outer ring volume by a radius Ro and Rc:
Vro = pi * (Ro^2 - Rc^2) .
The inner torus volume with help of r = **layer height**/2 is given by equation:
Vti = pi * r^2 * ( pi * Ri - 4/3 *r ) .
The outer torus volume by:
Vto = pi * r^2 * ( pi * Ro + 4/3 *r ) .
The needed radii are:
Rc = R + 1/2 * w
Ri = R + r
Ro = R + w - r 
with R the hole radius and w the extrusion width.
This assembles to an outer and inner volume:
Vo = Vro +Vto
Vi = Vri + Vti
The total extrude volume Vtot then simply still is: and half of it is a central reference volume Vc:
Vtot = Vi +Vo = 2 * Vc
and half of it is the further needed central reference volume Vc.
With this ingredients one can calculated an effective extrusion factor into the inner e_i and outer e_o side volume:
e_i = Vi / Vc
e_o = Vo / Vc
Using this equations with w = 0.4 and r = 0.1 leads to following plot of the effective extrusion factors above the hole radius:
[image]
The plot includes a further less detailed method, where the extruded volume is taken just as a torus with r = **extrusion width** /2,
and uses the same equations just exchanging Vi = Vti, Ri = Rc and Vo = Vto, Ro = Rc. 
Values for this method are marked as "by width". It is directly visible, that for smaller hole radius R the deviation between the inner and outer extrusion factors growe. This is obvious, since both sides recieve the same material volume, but have different geometrical volumes. Remarkable is, that the results for both methods are nearly the same. 

Now the Vi equation for both methods are inverted to Rc and Vc is taken as input volume. This corresponds to a move into the center of volume system resulting in variating Rc values needed to compensate a pure geometrical hole shrinking. 
The below plot shows the needed Rc values derived by both methods above the hole radius. 
[image]
Its visible, that the less detailed method leads to a constant Rc compensation over R. This originates from the fact, that the volumes Vi and Vo only differ by an added or subtracted term - 4/3 *r within the boundary condition Ri = Ro =Rc. Similar results were generated by the ArcCompensation algorithm, though this model always adds to an offset of half the extusion width. 
The results of the shown detailed model seem to be plausible, when comparing them to measure deviation values.

I like to summarize, that I am pretty sure there was no such method implemented in Slic3r or PrusaSlicer until the ArcCompesation was thrown out. Therefore this post should be taken as a feature request, since this is really something need to be done by the slicer.
Nevertheless the above mentioned hole compensation of SS, should be implemnted as well. Otherwise it is difficult to overcome printer deficits like z-wobbling or the technical need to fix the extrusion factor to a certain value. But please implement it as a material dependend parameter.

spiky2021, 2021-03-04:
I found a cut an past failure in my above text. It says "none wrong extrusion factor", but should be "wrong adjusted extrusion factor".

spiky2021, 2021-03-04:
Another failure! The second plot shows the difference between Rc = R + 1/2 w and the calculated one for geometrical hole compensation above the hole radius.

spiky2021, 2021-03-05:
I need to say sorry again. The equations for Vri and Vro miss the layerheigh h to make them Volumes.
Means:
Vri = pi * **h** *(Rc^2 - Ri^2)
and
Vro = pi * **h** *(Rc^2 - Ro^2)
The best would be to rewrite this thing?!

spiky2021, 2021-03-05:
Well, has to be:
Vro = pi * **h** *(Ro^2 - Rc^2)

spiky2021, 2021-03-05:
Hello again,
to confirm above model I printed two rulers with even and odd diameters (0,8mm - 10mm).
Done with I3 Mega S, calibrated half a year ago, stiff belt spanners, vibration compensation, 360° object cooling, filament Greentec Pro, infill 50%, 3 perimeters. Once they were printed with outer perimeter printed last and once first. For the other settings I used delivered PS standard settings to get results fast, maybe therefore edges are not as nice?? Bottom and top says extrusion factor 1 was OK. The used temperature was higher then needed (215°C) to make things worse due to lower viscosity.
The rulers are generated as high resolution .stl files with 10µm surface deviation and angel increments of 5°.
[image]
Then I measured each hole with a high precision dowel pin set having 10µm increments from 0.5mm to 5mm diameters and 50µm increments up to 10mm diameters. I started from quite low values to avoid streching the holes. 
[image]
Entering the results with and without printing outer perimeter first into the above shown plot of dR above R is shown below.
[image]
The lowest R values weren't measureable with the dowel pins, but here an 0.25mm and 0,3mm nozzel cleaning needle entered. 
A lower viscosity had no impact, but Greentec Pro anyway is less sensitive to temperature. 

A strange result, since above model fits well to the data measured with printing outer perimeter last and as well for
the data below 1.5mm diameter for printing it first, but else deviations nearly vanish!?
It behaves more like the old ArcCompensation model, but without its high offset??
Maybe this expert option is doing more than just printing perimeter first?
I like to investigate this further, since prints are not nice with printing outer perimeter first and the above model maybe able to compensate the shrinking with printing it last, as well.

If someone is interested I can send the excel file with calculations and data.

lonelymyp, 2021-03-06:
Titanic work, impressive.
For the sake of completeness, it would be interesting to measure the diameter of the holes in the g-code, but I do not know how this can be done. To compare how the slicing parameters affect.

spiky2021, 2021-03-08:
Hello again,
I searched in the PS code for the algorithms related to the two paramter **external_perimeter_first** and **small_perimeters_speed**.
In PerimeterGenerator.cpp activating **external_perimeter_first** just reverses of the vector holding all perimeters.
In GCode.cpp the speed under **small_perimeters_speed** is applied as soon a loop radius R is smaller than 6.8mm.
There is no other fancy stuff done.
For my prints shown before I recongnized, that the delivered default print settings for an I3 Mega sets **small_perimeters_speed** = 35 m/s, which is much to high for small holes. In general best would be to set it to 25% - 30% to include the need of printing very precise as well as very fast.
I printed the rulers again settings it to 15 m/s and entered it to my plot shown below.
[image]
This time I also printed rulers with T = 230°C to check the temperature dependency, but as visible it had not real effect.

It can be summerized, that the above shown model discribes the outcoming hole dimensions somehow, if no further advanced settings of PS are used.  As soon as  **external_perimeter_first** and a proper **small_perimeters_speed** is used the major deviations vanish. Though there still remains a significant deviation for diameters below 1mm. Stranges is, that by activating **external_perimeter_first** the deviation for diameters larger 4mm became larger. The rulers with this option on and off were printed several times at different bed positions and  infill types, but this systematic stayed. The roundness of these holes was Ok. There was no slit visible, when by pushing in a fitting dowel pin and holding it against the light. 

Due to this result I need to back my request to implement something like the above model, because PS is doing it right.
Rather it should be provided a parameter **small_hole_perimeters_first**. Or the **external_perimeter_first** parameters should
be exchanged by it, since i believe there is no other need for it, then for small holes. 
I know this implementation is more diffult. But maybe it works in PerimeterGenerator.cpp by somehow splitting the nesting of perimeters loops into the first contour, travers this splitted nests to extrusions entities afterwards, reverse only small hole entities and append all entities at the end? 

Due to my findings for holes larger 4mm radius, the actual small radius maybe not the right one to do so. On the other hand I believe there had been fundamental reason to set this radius to the precise value of 6,8mm? I will further investigate devitions for larger radii with the I3 Mega,  GreenTec pro and a caliper. 
By the way, can the reponsibles for the default I3 Mega settings be so kind to change **small_perimeters_speed** = 35 m/s for 0,2mm layer heights to a reasonbale setting and as well for other heigths?

spiky2021, 2021-03-08:
I like to extend above findings with some more thoughts. There can be envelop counturs (parents) of holes, that have a small diameter as well. There radius will be at least 1mm larger, otherwise both build a thin wall. Meaning this countours shrink less strong due to larger radius, shrinking is compenstated by the external hole perimeters sitting more outside after printing it first and the effect anyway is smaller for contours. Thus everything goes to the right direction!?

spiky2021, 2021-03-08:
Sorry for the many spelling mistakes, it was too much text for too less time!

astramateria, 2022-09-16:
More demand for this! Even Cura has this feature.

markmgit, 2022-12-31:
Cura 4.6.1 has "hole horizontal expansion".

mataksik, 2023-01-21:
everybody must optimize designs for slic3r hole error

sammcj, 2023-03-23:
Any update on adding this into PrusaSlicer? It's really useful in Cura.

SaschaHeyer, 2023-04-11:
https://www.youtube.com/watch?v=yzPqBt2SrcE

NVNDO, 2023-05-13:
Over one year later and STILL not solved... Grrrrrrr!!! 
https://github.com/prusa3d/PrusaSlicer/issues/5145

I just thought, lets give this PS another try, but NOPE, still cant get the holes right due to lack of functionality. 
I can't express in words how frustrating this is... So Cura it will remain.

Dids, 2023-05-13:
> Over one year later and STILL not solved... Grrrrrrr!!! #5145
> 
> I just thought, lets give this PS another try, but NOPE, still cant get the holes right due to lack of functionality. I can't express in words how frustrating this is... So Cura it will remain.

Tried SuperSlicer? It's effectively PrusaSlicer with more bells and whistles, including this particular feature.

spit40, 2023-07-03:
Please please add this. I run AuthentiKit which has a large community (>1,500) of people using 3d printing for replica flight simulation controls. Horizontal expansion settings are key to our printing (and design) workflow and I'd like to recommend Prusa Slicer to people instead of Cura as it is so much better for most things we need from a slicer (arranging multiple parts, placing an object on a specific face, selectively modifying objects for higher infill/walls etc and much more) BUT the horizontal expansion which applies to both perimeters and holes is a problem.  https://download.authentikit.org/index.php?members-list/

psi-4ward, 2023-08-02:
+1 e99999 :D

nerdCopter, 2023-08-25:
years 🤦‍♂️

ioma8, 2023-08-25:
Still waiting. :)

RandomGgames, 2023-08-25:
Still waiting :(

awabom, 2023-10-23:
I really want this too :-) 

(No, 'drill the holes' or 'fix it in cad' is worse than using Cura)

RandomGgames, 2023-10-24:
Problem is some models aren't yours so editing them isn't an option. Drill holes doesn't work either when it's an outside straight edge for example.

ChiefPoints, 2024-03-12:
Bumping the feature request, would be great to see this feature added.

spit40, 2024-03-13:
Quite right. Prusaslicer is getting all these upgrades. Surely this extra setting is worth some attention?

astramateria, 2024-03-14:
After a year regarding this I learned a lot. First off, I am still with Cura. Secondly, if you calibrate your flow rate just right, you really dont need to do expansion compensation a lot; I have mine set at less than 1 mm. Also I think other forks of PS (Orca?) have this feature too. So its not a big deal anymore.

ChiefPoints, 2024-03-14:
I believe using the flow rate to change the size of holes is a work around, not a solution. Now if you could change just the flow rate for holes, that might be a different story, otherwise the entire model gets the increased or decreased flow rates, which can affect print quality. The optimum solution is to be able to affect only the size of holes, which this feature request would accomplish. Now I do admit to using flow rate to affect these things, and in the meantime yours is good advice to try, but only because the necessary tool isn't there. One shouldn't have to affect the quality of the entire model just to make a hole smaller or larger, that's just my opinion, but it is shared by others. May not be a big deal to you, but it is to a lot of us. I've been fighting with a print lately that this feature request would of made simple, as it's easier than redesigning the entire model and sometimes all you have to work with is a .stl.

mooklepticon, 2024-09-07:
I'd like this feature, too.

NVNDO, 2024-10-12:
I have since turned this slicer my back as they don't listen to community request.

ioma8, 2024-10-12:
@NVNDO yea, PrusaSlicer devs do not listen to community requests at all. I too have had to switch to another slicer, whose devs care.

Tommy544, 2024-11-20:
+1 
Come on guys, this feature is clearly missed by many users. How difficult can it be to split the existing "XY size compensation" parameter into 2, allowing us to adjust shrinking of external dimensions and expanding of internal dimensions/holes separately?

astramateria, 2024-11-20:
I don't think it matters anymore considering that Orca Slicer, etc. have
incorporated this upstream. In fact, I switched to Orca and am not looking
back (used Cura). I still have Prusa in case I need to update my printer
etc.

MohrMa2, 2024-12-23:
Years wasted. I always fiddled with the xy compensation but failed with some objects. 
Just tried Orca slicers hole compensation and my calib matches. OMG. Damn, I liked prusa slicer.

borisburger, 2025-01-18:
It's interesting that Orca Slicer offers a feature to compensate for hole dimensions—this issue has been a constant frustration for me on PrusaSlicer. I'm planning to test OrcaSlicer to see how effective its compensation is.

From what I understand, fixing this isn't straightforward since it depends on nozzle size, filament material, perimeter walls, layer thickness, print speed, and cooling, and who knows what else. For example, with PET-G on my MK3S, a 0.6mm nozzle, and 3 perimeters, I use this equation to adjust hole sizes in CAD:

CAD diameter ~= 0.9 × Desired + 0.8

This is linear regression from test prints of holes ranging from 2.0mm to 6.5mm in 0.5mm increments. It works OK, though very small holes (<2mm) might behave differently.

I really like Autodesk Fusions' "Hole" feature, where you can insert say a hole for countersunk M4 bolt and specify that it is going to be tapped. Fusion knows 3.3mm is the recommended hole size for the M4 tap. But you can not use this feature with 3D printing, because the hole needs to be more like 3.77mm in CAD to achieve 3.3mm in print, so the predefined screw dimensions can not be taken as is.

MohrMa2, 2026-09-02:
Still a problem (switched to Orca, so not for me anymore).

Ioann44, 2026-09-05:
not legacy

lonelymyp, 2026-09-05:
> This issues has been labeled **legacy**.

not legacy!

nick-randal, 2026-09-06:
Use Orca

Wallby, 2026-09-06:
@Ioann44 @lonelymyp

This is expected from the bot, not a problem. Read this for information: https://github.com/prusa3d/PrusaSlicer/discussions/15593