prusa3d/PrusaSlicer #4562: Travel movements don't make sense

jhenley01, 2020-07-22:
### Version
2.2.0

### Operating system type + version
OSX Mojave

### 3D printer brand / version + firmware version (if known)
Prusa Mini 4.1

### Behavior

Hello,

I'm trying to minimize some of the blobs I'm getting on my PETG prints and I believe some of it is due to unnecessary or unoptimized travel movements. I've attached a screenshot of a print I do which includes a lot of small parts. 

When the print starts it draws the purge line and moves all the way to the back corner to start printing causing a small blob, why not start at the front? I work around this by having the purge line move from front to back but I still don't see why it doesn't start with the closest part at the front. Then it prints each object on the first line from left to right as expected but when it comes to the next row instead of dropping straight down to the next closest object it moves all the way to the far left and starts there.

Another thing I think could be improved upon is where the skirt starts and stops in relation to the part. Sometimes it will draw the skirt stopping at the front and then move to the back to start the part. Wouldn't it be better to have the skirt optimized to stop at the point where it would be closest to the part starting point?

To be clear the my retraction and flow rate is pretty well tuned so the blobbing is minimal but I think having better travel movements would definitely improve print quality. Any help or advice would be greatly appreciated.
[image]

jhenley01, 2020-08-27:
Glad I wasted my time with this post, guess I'll just live with this printer traveling all over the bed and leaving blobs for no apparent reason.

fionnb, 2024-10-02:
A video may be more illustrative than a thousand words...

https://github.com/user-attachments/assets/68b4009c-81b8-4959-8be1-d4d1e8d5d283

AndKe, 2026-09-18:
still unresolved:

[image]