prusa3d/PrusaSlicer #4564: New feature request: allow a single loop skirt with a multi-loop base

Xela-rev, 2020-07-23:
### Version
PrusaSlicer  2.2.0

### Operating system type + version
Windows 10 on Win64

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK2S

### New feature request
A single loop draft shield is a perfect protection against warping, but there is a problem for sticking that draft shield to the print bed: there is only one extrusion loop to hold the complete skirt to the print bed => warping can detach the skirt (while warping is exactly the reason to use that skirt).
If the skirt would have a fist layer with multiple loops, followed by only one loop for the rest of the height, the skirt would stick firmly to the print bed (while using a minimal amount of filament).

I propose to add an option that allows 2 different amounts for the skirt loops:
-> a number of loops in the first layer
-> a number of loops in the following layers (which must of course be smaller or equal to the value in the first layer)

In fact, this can also be interpreted as a brim, added to the skirt.

n8bot, 2020-07-23:
This could be achieved now if the height range modifiers would allow skirt/brim to be adjusted within them.

rtyr, 2020-07-23:
Skirt can intersect brim now, so you can use brim to increase skirt adhesion to the bed. See also https://github.com/prusa3d/PrusaSlicer/issues/724.

[image]

n8bot, 2020-07-23:
> Skirt can intersect the brim now (brim extrusions are split and a gap for the skirt is inserted) [...]

It might be useful to not split the brim, print it whole, and split the skirt instead on layers that they overlap, no?

wavexx, 2021-08-14:
> It might be useful to not split the brim, print it whole, and split the skirt instead on layers that they overlap, no?

I fully agree. See #6801 (old #3692)

n8bot, 2021-11-13:
I really think this is important.

We need to be able to specifically place the "skirt" onto a raft. Or, provide a brim for the skirt. Because the skirt can be printed really tall, and perhaps using multiple extruders, which is a nightmare for the skirt warping. We need control over the first layers of adhesion for the skirt!

And, I do not use "draft shield" because IIRC, it also changes the heating. tool changing behaviour. From what I can tell, the brim intersected by the skirt is disabled if this feature is disabled.

The hardcoded settings sometimes force the hand too much of the user, while still providing lots of ways to make non-sensical setting combinations. Why force our hands so much?

closed 2026-09-29 by prusaslicer-bot[bot]