prusa3d/PrusaSlicer #4566: Automatic scaling........ With NO CHOICE

Jonathanp19, 2020-07-24:
Prusa slicer 2.2.0 

Importing large prints to be sliced in the slicer into smaller chunks, AM unable to as each time i do your sliver FORCES the scale onto the object to a random scale and says that its the new 100% . very annoying. Even though the scale is locked your software forces the change ans also forces the change if you click the X on the pop up.

lukasmatena, 2020-07-24:
What is the size of your bed and the model? Could you send your config and model? As far as I know, the automatic rescaling of the mesh is only applied in extreme cases where the object is so big that it might trigger fixed-point arithmetics overflows.

rtyr, 2021-04-22:
No response. Closing.

closed 2021-04-22 by rtyr