prusa3d/PrusaSlicer #4567: custom bed logo / STL files are not copied

belm0, 2020-07-24:
### Version
2.2.0+

### Operating system type + version
OS X

### Behavior
* _Describe the problem_ When setting a custom bed logo/stl, PrusaSlicer only saves a reference to the files.  If the files are later deleted (for example, user deleted his downloaded files), the custom bed display will be lost.   PrusaSlicer should copy the given files to a location where its other configuration files are stored.
* _Steps needed to reproduce the problem_
    1. in Printer Settings / General, set custom png or stl on bed shape
    2. close PrusaSlicer
    3. delete the png or stl file
    4. open PrusaSlicer
* _Expected Results_: custom bed shape is displayed.
* _Actual Results_: default bed shape is displayed

_Is this a new feature request?_ No

foreachthing, 2020-07-24:
Save your png/stl to your profile folder...

JohnOCFII, 2020-07-24:
Related to this -- I maintain a profile for the Railcore 300 printer, and I'd like to distribute bed SVG/STLs with it, but right now it is a separate step to install them, and location varies based on OS.  It would be nice it it could be bundled and that the PrusaSlicer installer could move them to the appropriate spot.

belm0, 2020-07-25:
> Save your png/stl to your profile folder and get them from there...

99% of users configuring the bed customization are downloading 3rd party profiles.  They do not know this detail, nor where the "profile folder" is on their particular OS.

The info could be included in the Prusa docs, and hope profile developers read it, and hope they in turn include the detail in their profile instructions (including where the profiles folder might be on any particular OS), and hope their users will follow that correctly...

Rather than the current unfriendly experience, any of these would be the right thing to do (from most preferable to least):
   1. make it possible to include bed stl / png as part of the profile bundle (suggested by @JohnOCFII)
   2. have PrusaSlicer copy the files selected in the UI to its configuration folder
   3. have the custom bed UI display the configured file paths.  Then it's more obvious that the configuration only stores a path reference, and users can see what the current path is.

foreachthing, 2020-07-25:
> 2. have PrusaSlicer copy the files selected in the UI to its configuration folder

I agree. I had to learn the hard way, too.

closed 2026-09-29 by prusaslicer-bot[bot]