prusa3d/PrusaSlicer #4568: Feature Request: Add Disk as primitive mouse ears

daktylus, 2020-07-24:
Sometimes it is extremely helpful to add mouse ears to  your print to prevent lifting of corners due to warping. This can be achieved in PrusaSlicer by adding the primitive cylinder to your objec, scaling and correct positioning. It would be great if one could directly choose Add part > Disk and one would get a Disk with dimensions of about 0.2 mm height and 10-20 mm diameter. Should be fairly easy to implement, but extremely usefull

[image]

[image]

AdamLeyshon, 2020-08-20:
Agreed, I think this would be more efficient than printing a brim for the whole part for parts that you know don't need brims but the corners still curl.

wnovinger, 2020-08-29:
+1 here. I have wanted something similar on several occasions.

al3ph, 2020-12-30:
+1 would be very handy, also good if it was invisible to the auto layout code.

Kachidoki2807, 2020-12-31:
It would be even nicer if it can be auto-placed with a simple click on a vertex of the face onto the plater.
But it's becoming a new feature, and not an add to the primitive shape feature anymore.

BHoggs, 2021-11-23:
SuperSlicer has a nice implementation of this:
[image]

Result:
[image]

neilmendoza, 2023-05-31:
Big +1 to having similar brim ear functionality to Super Slicer. This is the biggest missing feature for me in Prusa Slicer at the moment.

AustinTylerDean, 2023-06-25:
GIANT +1 to adding the already added in SuperSlicer brim ears!

viesturz, 2023-09-13:
+1 please add this.

Even better if you can make auto generation option for all corners where angle > x.

lorenzofanchi, 2023-10-01:
+1, just lift it from SuperSlicer (credit where due ofc) 😄

wikolii, 2023-10-03:
+1 please add this. Super helpful for those who always do this manually for PETG prints

DaTTcz, 2023-10-29:
Když jsou uši 4 tak si je tam dám ručně, ale osazovat ručně třeba 10 a víc kruhů u složitějšího modelu už je docela opruz. Mám pocit, že si tuhle funkci PrusaSlicer zaslouží.

belyaev-ban, 2023-12-05:
+1. This would be a very useful feature and save lots of time!

Snuff1eupagus, 2024-05-03:
Update: This has been possible to do in PrusaSlicer since Sept. 1, 2021.

githubniko, 2024-05-05:
Still not done?

tk-notify, 2024-05-20:
Would be great to have the brim ear option. The mentioned work arounds are not working for me. Looking forward to an implementation.
Thanks!

closed 2026-09-29 by prusaslicer-bot[bot]