prusa3d/PrusaSlicer #4575: Detect Thin Walls causes holes

rennne, 2020-07-26:
PrusaSlicer is 2.2 on Windows 10
0,3mm layer height and 0,6mm Prusa Nozlle Profile.

When I use "Detect Thin Walls" the printing time drops from over 5 hours less than 4 hours -> this is good

but it causes holes in the object -> this is not good
The holes are not there is I do not use this feature

[image]

hte zip-stl-file:
[lampenschirmV4.zip](https://github.com/prusa3d/PrusaSlicer/files/4977959/lampenschirmV4.zip)

rtyr, 2020-07-26:
Similar to #340, #149, #3611. #4110, #2583

rennne, 2020-07-27:
It is a bit different in that regard, that it produces holes in places where the wall section of that layer is more than one millimeter.

So it causes see through holes in "thick" potions of the wall

PaparazziN, 2023-10-22:
Hi @rennne!
Have you tried using the 'new' Arachne Perimeter generator? I tried it with the file you attached and it comes out a lot better with the Arachne compared to the Classic generator. Based on the similarities with other issues and the age, it might be worth considering closing this issue.

closed 2026-09-29 by prusaslicer-bot[bot]